Denvi/Candle #164: Проблема с визуализатором (?)

inFamousMcGrath, 2017-09-02:
Я управляю станком с помощью Arduino Mega 2560, с прошивкой 1.1f с репозитория grbl-Mega. Когда я запускаю на выполнение код, первое время всё идет хорошо - визуализатор показывает всё как надо, работает панель изменения подачи, скорости и оборотов и т.д. Но через некоторое время происходит что то странное - программа продолжает отправлять код, но визуализатор не двигается, меню не работают, координаты в верхнем правом углу замирают. но станок работает, и самое интересное, что вполне корректно, строки продолжают отправляться программой (работает автопрокрутка и статус строки кода), никаких ошибок, никаких уведомлений нет.

Peter6891, 2017-09-04:
у меня подобная ситуация. только мк ATMega328. прошивка grbl 0.9j или 1.1f
периодически мигает красным статус ("Неизвестно)", а иногда просто статус горит зелёным ("Готов")
но через короткое время замирает программа, и чуть позже и контроллер замирает
...

inFamousMcGrath, 2017-09-04:
Я определил проблему, поскольку пишу свою собственную программу (больше похожую на промышленную стойку). Всё дело в ошибках передачи. Эта программа никак не реагирует на ошибки протокола USB, которые есть на Atmega16, который стоит как UART преобразователь на Mega2560. Поэтому сначала пропадает визуализация, а позже и перестает работать сама отправка файлов.

Peter6891, 2017-09-04:
у меня преобразователь не на меге, а на CH340G

Peter6891, 2017-09-04:
подбор кварца с дробной частотой может решить проблему?
по рекомендации стоят на 12 и 16 МГц. может есть смысл подобрать с дробной частотой чтоб уменьшить потери данных передачи из-за не стыковок частот?

Peter6891, 2017-09-04:
[image]

inFamousMcGrath, 2017-09-04:
Перекомпилируйте Grbl с baudrate 9600, пишут что помогает от потерь на CH340G

Peter6891, 2017-09-04:
я вас немного опередил)) перекомпилировал именно на 9600 и 2 часа без перебоя отработал

Peter6891, 2017-09-04:
спасибо.

closed 2017-09-21 by inFamousMcGrath

zdanat, 2018-07-17:
Уважаемый Peter6891, можете ли Вы поделиться перекомпиляции версией, я к сожалению не владею навыками подобных действий, а у меня возникла такая же проблема со станком. Спасибо.

Peter6891, 2018-07-18:
а какие параметры именно вам нужны? у вас ATMega 328 ?
версия grbl какая?

zdanat, 2018-07-18:
Grbl 0.9j и ATMega 328, все симптомы проблем как описаны в первом посту

Peter6891, 2018-07-18:
[grbl.zip](https://github.com/Denvi/Candle/files/2205295/grbl.zip)
попробуйте

zdanat, 2018-07-18:
Большое благодарю, буду пробовать

Peter6891, 2018-07-18:
попробуйте так:

[grbl.zip](https://github.com/Denvi/Candle/files/2205305/grbl.zip)

Peter6891, 2018-07-18:
в предыдущем (которое удалил) я забыл добавить файлы прошивки. в этом добавил.