Denvi/Candle #167: Не запускается с официальными драйверами для Intel GMA 3150

inFamousMcGrath, 2017-09-21:
При установке официальных драйверов перестает запускаться. Ошибка стандартная Windows - Candle прекратил работу. Причем, если удалить драйвера и сидеть на драйвере который предлагает Windows - запускается, даже работает визуализатор(!). Жаль что ноутбуком только нормально пользоваться нельзя, а так работает.

Denvi, 2017-10-05:
Запустите [GPU Caps Viewer](http://www.ozone3d.net/gpu_caps_viewer/), что там в графе "OpenGL" с драйверами от Intel/Windows?

inFamousMcGrath, 2017-10-06:
Найду время посмотреть - скажу, потому что я нашел оригинальный способ запустить его без визуализатора на офф. драйверах - просто удалил файл OpenGL от Интел, название уже не помню, оно было в подробностях ошибки, и всё работает, визуализатор мне и не нужен, вообщем то)

closed 2017-12-28 by Denvi