Denvi/Candle #169: Z щуп  при команде M6

yuranikspb, 2017-09-21:
Создаю gcode в программе FlatCAM
при сверление отверстий ставлю галочку toolchange и в g-code добавляется команда M6
хотелось бы чтобы c Candle могла реагировать на этот код 
например пользовательской командой ( к примеру отключить шпиндель уйти в x0 y0 z20 поставить на паузу) а дальше команда z- щуп и продолжение основного g кода.
проблема в том что при смене сверла нудна заново мерять высоту.

Denvi, 2017-10-05:
Действительно, хотелось бы. Запишу в список функций на реализацию.

grblkit, 2017-10-18:
У меня на M6 ошибка
Приходится М05 и М00 использовать, а дальше кнопку старт жать. grbl 1,1f
candle 1.1.7
flatcam 8.5

AlexeyGusev, 2017-12-07:
плюс 1 голос за реализацию, Candle ругается на M06. Игловский pcb-gcode генерирует g-код с командой M06 для смены фрез/свёрел, если ему указать drl-файл (список доступных инструментов). Если бы данную функцию ещё и сбэкпортить в ветку 1.0 для grbl 0.9, тогда было бы вообще круто.

Ramzezka, 2017-12-24:
+1
Собрал станок, начал работать c GRBL для сверловки плат. Команды смены инструмента вызывают ошибки. Их конечно можно заигнорить, но тогда не будет паузы для смены инструмента.

Пришлось заигнорить и ручками в гкод вставить М00 для паузы и смены.

Но вот беда - во время паузы нельзя управлять с компа станком или сделать Z щуп, а сверло то другого размера.

Поэтому на паузе приходится руками крутить за винт оси Z , что бы регулировать в ноль ))

Идеально было бы на команду смены инструмента поднимать на высоту, заданную где то в настройках и делать паузу. Идеально Х2 если бы была разблокирована кнопка Z щупа, что бы новую фрезу или сверло выровнять.