Denvi/Candle #170: CNC machine cutting projects very small

shabbee, 2017-09-23:
I'm a complete newbie, apologies in advance for being dumb.

I've just started tested my Tomtop mini cnc but everything I cut is tiny. I've used a number of source files so don't think the problem is there. I've not messed around with any settings apart from the jog values. Do these need to be set to a specific value for the software to cut 1:1?
Thanks all

nojahoda, 2017-10-02:
Hi i had the same issue, after 24 hour of testing, i found a work around.

I use FLATCAM to create the gcode.

1. On Flatcam load the gerber file in INCH, then change it to mm.
2. Zero your pcb
3. Scale it by 3.2
4. export your gcode
5 upload it to candle

3.2 is the magic number 

I am also a newbie

Denvi, 2017-10-05:
Hello, looks like you have wrong "steps/mm" setting in firmware.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

>#### $100, $101 and $102 – [X,Y,Z] steps/mm

>Grbl needs to know how far each step will take the tool in reality. To calculate steps/mm for an axis of >your machine you need to know:

>* The mm traveled per revolution of your stepper motor. This is dependent on your belt drive gears or lead screw pitch.
>* The full steps per revolution of your steppers (typically 200)
>* The microsteps per step of your controller (typically 1, 2, 4, 8, or 16). _Tip: Using high microstep values (e.g., 16) can reduce your stepper motor torque, so use the lowest that gives you the desired axis resolution and comfortable running properties._

>The steps/mm can then be calculated like this: ```steps_per_mm = (steps_per_revolution*microsteps)/mm_per_rev```

>Compute this value for every axis and write these settings to Grbl.

nojahoda, 2017-10-06:
Indeed i have wrong setting.
I managed to change them  and now everything works fine.

Thank you

shabbee, 2017-10-08:
Thanks guys, all sorted now.

closed 2025-09-02 by github-actions[bot]