Denvi/Candle #172: Soft Start for M3

yamimann, 2017-10-03:
Hello,
i use a 2417 CNC with a new spindle motor. It would be fine if i can use a soft start (by software) for this motor because the start power consumption is very high and my PSU stopped. If i start the motor with 75% everything is fine but i have to speed up the motor manually. Is it possible to implement a routine to start the motor e.g. at 15% and then increasing to 100% in 2-3 seconds?

regards
Joerg

Denvi, 2017-12-28:
GRBL firmware relative problem.

closed 2017-12-28 by Denvi

pedjas, 2020-09-26:
SaintSmart released new stronger spindle motor for their CNC 3018 machine. Lots of people are reporting issues using it. It takes too much power to start that causes various issues. If started low speed and gradually speed up, it works fine.

Having option for soft start would resolve most of their problems.