Denvi/Candle #173: Win 10 64-Bit - App does not run

abhidatyahoocom, 2017-10-07:
I have tried V0.8, 0.9 GrblControl and 1.0 Candlelight - It only runs once or twice and never again.  Unless I move to a lower or higher version of the software and then it runs just once or twice.

Is Win 10 64-bit supported?

Thanks,
Abhi

Denvi, 2017-10-07:
Hello, post settings.ini file of non working program.
Win 10 x64 supported.

abhidatyahoocom, 2017-10-07:
Attached, thank you.....
[settings.ini.txt](https://github.com/Denvi/Candle/files/1366319/settings.ini.txt)

biasedlogic, 2017-10-09:
I observe similar behaviour, short time after failed restart of the software the whole system comes down with memory management error. The PC is otherwise stable with uptimes upwards of a month...

Denvi, 2017-10-10:
@abhidatyahoocom 

Closing "Candle" window in non-main screen in multiscreen setup will cause this bug.

So:
- open "settings.ini" in text editor;
- remove line, which starts with "formGeometry=@ByteArray";
- save changes;
- start "Candle".

To avoid bug, please, close "Candle" when window is on your main screen.

abhidatyahoocom, 2017-10-11:
@Denvi 

Thanks very much !!!  Working fine now.....

Abhi

azmanishak, 2020-02-01:
Thank you so much.

> @abhidatyahoocom
>
> Closing "Candle" window in non-main screen in multiscreen setup will cause this bug.
>
> So:
>
> * open "settings.ini" in text editor;
> * remove line, which starts with [...]

glkeeney1951, 2020-09-13:
There's an easier way:
* Download the software and place it in a folder in "Documents".
* Navigate to the "settings.ini" file and open in notepad.
* Delete the "formGeometry=@bytearray" line. (as above)
* Save the modified settings.ini file.
* Copy and paste the GRBL software folder (in my case grblControl0.8) in Local Disk (C:) > Program Files (x86).
* Create a Shortcut for the "grblControl.exe" file and paste it on you desktop.
* Double click on the desktop icon and grblControl (or Candle) should launch.
* Remember NOT to open or drag the software to a screen other than screen 1.

jholovacs, 2020-12-10:
...that's an _easier_ way?  I'm also quite annoyed by this, and no matter if I follow these instructions, I'm still told that it won't work with windows, then there are a bunch of other DLLs it can't load and it ultimately crashes.

DEF7, 2021-01-04:
@jholovacs

> ...that's an _easier_ way? I'm also quite annoyed by this, and no matter if I follow these instructions, I'm still told that it won't work with windows, then there are a bunch of other DLLs it can't load and it ultimately crashes.

Being annoyed isn't going to fix anything - the developer doesn't owe you anything. When you use free software you can either meet the developer halfway and figure out how to make it work by putting some effort into it, or you can try your luck elsewhere.

It works fine with Windows. I've only ever used grblControl/Candle with Windows. The point of the fix is that you apparently can't use it with a multi-monitor setup unless you only have Candle running on your primary monitor and when you close it you must directly close the window itself rather than having it in the background and right-clicking it in the taskbar to close it (from what I gather).

It sounds like you have an unusual Windows setup. Are you sure you're not trying to run a 64-bit program on a 32-bit version of Windows? Have you tried installing it in a user folder where Windows will give it access to its own files, or running as administrator? These are things you could be doing instead of telling the internet you're annoyed by completely free open source software that doesn't owe you anything, at all.

jholovacs, 2021-01-04:
Can we forego the sanctimonious speeches please?  I'm fully aware an open source developer doesn't owe me anything, but I don't have an unusual setup, this is the software that's included in the box of a large number of CNC routers (including mine), and the primary thing software needs to do is _work_; bells, whistles, and polish notwithstanding.  What serious computer user doesn't have a multi-monitor setup nowadays?  (that being said, this latest version literally does not work on my laptop straight from the factory.)  As a professional technologist who's been in software development his whole life, contributed to many open-source projects, and also leads his own development team, I have a hard time considering this even at an alpha stage, and I think I'm justified in my annoyance at what is generally considered the "standard" in Grbl software available being this flaky.

To answer your questions: both my desktop and laptop are running Windows 10 Home x64.  The desktop is dual-monitor but the laptop is not.  I have attempted to run as administrator, though I haven't tried to set permissions on the file system; I will try that tonight.  I feel quite justified in my annoyance, and IMO you could be spending your time more effectively by not attempting to lecture people who work in software development about what they are justified to be annoyed about regarding open source code, or by expressing their annoyance.

DEF7, 2021-01-04:
As someone who accuses others of being sanctimonious it's scary knowing people like you exist when you then go and say things like:

> As a professional technologist who's been in software development his whole life, contributed to many open-source projects, and also leads his own development team

Whatever you say, hotshot. I mean, you're clearly the most experienced and qualified person here to have an opinion just because you make the mistake of thinking so highly of yourself, right? That's rich.

> What _serious computer user_ doesn't have a multi-monitor setup nowadays?

I guess anybody cooking with one monitor can't be a hotshot like you, even though there are invariably countless developers who are more skilled, knowledgeable, and emotionally mature than you who are only running one monitor out there in the great wide world.

> I have a hard time considering this even at an alpha stage

For claiming to be such a hotshot you sure are saying a lot of things that don't jive with that claim. At least in this one small instance we can get you squared away so you don't repeat it in the future, you're welcome: https://bfy.tw/Q3eL

> I think I'm justified in my annoyance at what is generally considered the "standard" in Grbl software available being this flaky.

I've never heard of Candle being the standard, and I've been doing the hobby CNC thing for years. Just because some morons packaged it with their CNC hardware  doesn't mean it's what everyone uses - there are plenty of other programs that people have been using since before Candle even existed (i.e. UGS, . This is just a FOSS program some random dude was nice enough to share with the rest of the world in the event that they stood to benefit from it in any way possible. There was no guarantee it was going to be worth anything to anybody else, nobody promised that (except maybe the people you made the mistake of buying a CNC from). Therefore, you're not justified in your annoyance because it's founded on your sense of entitlement, which should be directed at the people who took your money.

> my desktop and laptop are running Windows 10 Home x64

Not that this is relevant to the issue you're having, but I thought you were a "serious" computer user, and you're running Windows Home? The hotshot facade continues disintegrating before our very eyes.

> you could be spending your time more effectively by not attempting to lecture people who work in software development

You really need to stop assuming you're the only person who makes a living writing code and waving it around like it somehow absolves you of the responsibility that every human being has to not be an egotistical lame toward their fellow humans. For someone who waves around the claim that they're a "professional" your behavior betrays the fact that you're anything but.

jholovacs, 2021-01-04:
@DEF7 You seem to have lost your mind.  Not reading your diatribe, and shame on you for turning a github issue into some youtube rant with personal attacks.  Let me know when you actually have some value to add to the issue, and I'll read it.

jholovacs, 2021-01-04:
@Braunfeltd my problem is I can't even get the sw to start up; errors with various dependency libraries, and not the ones I'm used to (like 32/64 format or wrong version stuff)... as mentioned above I've tried with my main desktop and a laptop I bought expressly for working with my new CNC router, so I can accept I'm missing something but the problem is not obvious at all.

Braunfeltd, 2021-01-04:
for mine, I download the beta zip file.   unzip into candle folder on desktop and just run candle.exe  I have different versions as I was testing each in a different folder.   the DLL's should all be in the same folder along with the rest of the candle directory structure.   Could be something else missing.   Are the machines all windows patched up?  I run mine on a windows 10 pro really old tablet from Samsung which works, as well I ran it on two other laptops one on Home and the other pro as well.    when does it crash exactly?  when you are loading a file or soon as you click on the executable?

DEF7, 2021-01-05:
Sounds like the classic "running from the zip file" problem.

DEF7, 2021-01-05:
[image]

If this is the kind of thing you're seeing it's because you're trying to run it from the zip file, which is a very common thing for computer novices to do - I have people do it all the time with the software I write and sell.

Riot54, 2021-04-08:
I know this is old but for the sake of google searches: 

If the above fixes aren't working, run the windows 10 compatibility troubleshooter, it should reset the window to your primary monitor, this is how I fixed the issue for myself.

blounolito, 2022-07-25:
I used to encounter this weird problem, quick fix is to open Candle / grblControl from your **main** screen.
Works like a charm.