Denvi/Candle #174: Add release for OSX 64bits

Nandox7, 2017-10-18:
Gave this a try and it compile and runs without much hassle.
Downloaded https://download.qt.io/archive/qt/5.4/5.4.2/qt-opensource-mac-x64-clang-5.4.2.dmg and installed it full with all addons (to make sure the serial addon is present)

And only had to make a small adjustment to QT to allow it to work correctly with current version of OSX. As detailed here: https://stackoverflow.com/questions/33728905/qt-creator-project-error-xcode-not-set-up-properly-you-may-need-to-confirm-t

Issues Found:
- Colors are not set as in windows versions and it is using black color for all.

se3d, 2017-10-20:
yes.... a good request.

Nandox7, 2018-01-05:
Gave this a try and managed to pack it for OSX.
I have built it in OSX 10.12 so I suspect at the moment will only run on this version or higher.
Code is for version 1.1.8 that is currently under branch grbl_1_1.

https://www.dropbox.com/s/31rrx9n7musvz8d/Candle_1.1.8.dmg?dl=0

tombs007, 2019-03-30:
@Nandox7 Do you know if this works with HIgh Sierra(10.13.6)? Candle starts but I doesn't detect my device. I can manage my router with easel and bCNC fwiw.

piersg, 2019-06-09:
1. HUGE thanks to @Nandox7 for saving me a download/compile/tedium (I had 500M of qt downloaded when I found this build)
2. @tombs007 Works for me on High Sierra as long as: I prepend "/dev/cu." to se serial port name (caveat: I haven't actually milled anything yet, but it's connecting).

E.g. I'm connecting to `/dev/cu.wchusbserial1420`

Nandox7, 2019-06-10:
Appologies @tombs007 I had missed you previous question but @piersg already
replied to you.

I see that there are a few updates since my last build, so I'll try and
compile a new version to include the latest commits.

-Nando

ah1102, 2020-02-01:
macOS Catalina not work

vol4enok, 2020-08-31:
I want to share an assembly of Candle for macOS. Tested latest build version 1.2.9.1b on MacOS Catalina (10.15.6) and it works well. You can always download the dmg files from the [link](https://drive.google.com/drive/folders/1AGgX804tFUevi9AKxamSViYKNyABAJ_W?usp=sharing)

eightheads, 2021-02-05:
> I want to share an assembly of Candle for macOS. Tested latest build version 1.2.9.1b on MacOS Catalina (10.15.6) and it works well. You can always download the dmg files from the [link](https://drive.google.com/drive/folders/1AGgX804tFUevi9AKxamSViYKNyABAJ_W?usp=sharing)

Thanks for this!

Nandox7, 2021-08-18:
So had to go back at this.
Got it to build to Big Sur with latest LTS QT version 5.12.11, more details to be added soon.

vol4enok, 2021-08-18:
> I want to share an assembly of Candle for macOS. Tested latest build version 1.2.9.1b on MacOS Catalina (10.15.6) and it works well. You can always download the dmg files from the [link](https://drive.google.com/drive/folders/1AGgX804tFUevi9AKxamSViYKNyABAJ_W?usp=sharing)

New builds 1.2.12b and 1.2.14b available.

synman, 2021-12-20:
any builds available for Monterey?   I noticed Candle2 (liked above by @vol4enok crashes on startup.

kpbicka, 2023-11-03:
> I want to share an assembly of Candle for macOS. Tested latest build version 1.2.9.1b on MacOS Catalina (10.15.6) and it works well. You can always download the dmg files from the [link](https://drive.google.com/drive/folders/1AGgX804tFUevi9AKxamSViYKNyABAJ_W?usp=sharing)

Коллега, это же Ваш репозиторий с разными версиями Candle?
Поскольку направление CNC  у Вас, нет ли, например собранного приложения GERBER_X3 под Мак?

vol4enok, 2023-11-07:
> Коллега, это же Ваш репозиторий с разными версиями Candle? Поскольку направление CNC у Вас, нет ли, например собранного приложения GERBER_X3 под Мак?

К сожалению нет. Пользуюсь FlatCAM для этих целей. Под Мак работает отлично.

kpbicka, 2023-11-07:
> > Коллега, это же Ваш репозиторий с разными версиями Candle? Поскольку направление CNC у Вас, нет ли, например собранного приложения GERBER_X3 под Мак?
> 
> К сожалению нет. Пользуюсь FlatCAM для этих целей. Под Мак работает отлично.

FlatCam тоже использую - нареканий нет. А LightBurn нет под мак?)

vol4enok, 2023-11-07:
> > > Коллега, это же Ваш репозиторий с разными версиями Candle? Поскольку направление CNC у Вас, нет ли, например собранного приложения GERBER_X3 под Мак?
> > 
> > 
> > К сожалению нет. Пользуюсь FlatCAM для этих целей. Под Мак работает отлично.
> 
> FlatCam тоже использую - нареканий нет. А LightBurn нет под мак?)

не пользую.

thirstyone, 2024-03-25:
nice recent files ;)

closed 2025-09-02 by github-actions[bot]