Denvi/Candle #178: Настройки карты высот неактивны

dmg244, 2017-11-11:
Пожалуйста подскажите, Candle 1.1.7 + Win7 64bit
При создании карты высот, настройки серые т.е. неактивны. Соответственно невозможно выбрать область и т.д. 
[image]
Как их активировать?

Denvi, 2017-11-11:
На панели "Перемещение" снять флаг "Управление с клавиатуры".

dmg244, 2017-11-11:
Действительно! Благодарю, никогда бы не подумал! А почему так, с чем это связано?

Denvi, 2017-11-12:
Для исключения двойственности управления: цифровые клавиши, нажатые при установленном фокусе в той же таблице команд, приведут при нажатии к вводу цифр в таблицу, а не к управлению станком.

dmg244, 2017-11-12:
Благодарю!

aspuz, 2022-06-01:
здравствуйте. я никак не могу разобраться с картой высот. пытался с разными версиями CANDLE, на всех одинаково. стол, относительно, ровный.
[image]
версия grbl 0.9. пожалуйста помогите настроить карту высот.

mar0x, 2022-06-06:
Здравствуйте.
Попробуйте переместить шпиндель в точки, которые использовались для построения карты высот.
Например: в точке (0, 10.42) высота определилась 2.56 мм, перемещаем шпиндель на 0.1 мм выше этой точки: "G90 G0 X0 Y10.42 Z2.66" - теперь смотрите где он остановился, если в 0.1 мм над поверхностью - значит такой стол, если сильно выше, значит надо искать проблему в щупе или процессе построения карты высот.

closed 2025-09-02 by github-actions[bot]