Denvi/Candle #182: Processing size problem ...

yunocho, 2017-12-02:
grbl Version 8.4 is in use. I am using the cnc3018 model.

When using a spindle motor, it is machined 5 times larger than the dimension in the .nc file. For example, if x = 30mm, y = 10mm, z = 10mm, the x and y values ​​will be machined to the sizes of 150mm and 50mm.

Is there any way I can figure out how to set it up? If you know how to fix the problem, please let me know. Please.﻿

ASHIT-AXAR, 2017-12-02:
you have done wrong step/mm calculations.
post your settings here (typing $$ in the console)

yunocho, 2017-12-03:
Serial port error 1: No such file or directory
[CTRL+X] < Grbl 0.9j ['$' for help]
$$ < $0=10 (step pulse, usec)
$1=200 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=4 (dir port invert mask:00000100)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$80=2 (diy8000 data1,int)
$81=14 (diy8000 data2,int)
$82=0 (diy8000 data3,int)
$83=0 (diy8000 data4,int)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=20.000 (x accel, mm/sec^2)
$121=20.000 (y accel, mm/sec^2)
$122=20.000 (z accel, mm/sec^2)
$130=2000.000 (x max travel, mm)
$131=2000.000 (y max travel, mm)
$132=2000.000 (z max travel, mm)

ASHIT-AXAR, 2017-12-04:
try setting following values:
$100=160
$101=160

yunocho, 2017-12-05:
We have solved the difficult problem!
One question is how do you calculate the $ 100 and $ 101 settings?
I wonder why I should give a value of 160.

ASHIT-AXAR, 2017-12-05:
we don't.
The Gods have provided these instruments and settings for us.
the profound temple high in the heavens
turned into profound lightning
the brilliant power of the heavens
that can cross over the horizons
and reach into our hands
let it be used wisely ...

closed 2017-12-06 by yunocho