Denvi/Candle #183: Управление клавиатурой.

Nezarod, 2017-12-04:
Добрый день.
Искал в документации - но не нашел.
Какие кнопки задействованы при управлении с клавиатуры?
Есть где то описание?
И можно ли их переназначить?
Спасибо!

Denvi, 2017-12-04:
Для версий 1.0.х:

[image]

'Num 0' - вкл./выкл. шпиндель;
'Num *', 'Num /' - задать скорость вращения шпинделя.

В версиях 1.1.х внесены изменения:
'Num 5' - остановка перемещения;
'Num +', 'Num - ' - выбор подачи;
'Num 7', 'Num 1' - выбор шага.

Функции переназначения клавиш на данный момент нет.

Nezarod, 2017-12-04:
Спасибо!
А вызвать запрограммированные макросы, которые на кнопках, 1-2-3-4, с клавиатуры никак нельзя?

Denvi, 2017-12-28:
На данный момент нельзя.

Kearvan, 2020-05-26:
Жаль. Есть ли в планах добавление клавиатурных комбинаций для панелей "Управление" и "Команды"?