Denvi/Candle #185: Resetting machine coordinates and a couple more.

Larepera, 2017-12-12:
Hi all!
I am testing Candle 1.1.7... It is really nice and performs quite well, but I would like to report some issues that I have detected along my tests. I am using VisualMill (actually VisualCad 2012), original (and paid) version. This software has an internal post-processor editor witch was very helpful to tune up a post-processor for GRBL in a way to end with a mechanizing process without error or interruptions.

One of the issues detected is that, after the end of the process, I cannot move the machine to its 0,0,0 position. When I click on the Origin key (icon?) I get the following answer:

$H < error:5
S10000 < ok
Not move at all.

If I click on the key behind Home (the one with the house icon) I get the following answer:

G21 < ok
G53G90G0X287Y-178.3Z0 < ok
G92X287Y-178.3Z0 < ok
But nothing happen.
I cannot find the G92 code in the list of codes supported by GRBL. May be the problem is around this.

Other issue is that it seems that Candle do not like the Comments very much. If I leave the comments in the CNC program, Candle terminate the execution way before actually ending with the program but saying "End Job" or something like that (I have the software installed in Spanish) and the time it took to this point. As you can see in the picture, it is way too much before the real end of the job.
If I send the same program but without the Comments, it will be OK finding the end of the job without problems.

Nest thing is just a suggestion. I enabled the Overriding function and set it at 200% because I was not actually mechanizing but wanted to run as fast as possible. That worked great but what I would like to suggest is that it would be better to dis-enable this function after the job is completed. Currently this option remain active with the last setting so you can face problems in the case of forgetting setting it back to 100%.

Finally, I would like to comment that the program (Candle) was set automatically to Spanish because my PC settings. That is OK, but I can not find a way to change the language to English. There are some text without translation and also many of them that are not so properly translated. It would be nice to have the possibility to set the language manually.

Well, that's all. Thank you for your attention and the great software!

[image]

Denvi, 2017-12-28:
Hello

> $H < error:5

Please check GRBL firmware documentation. https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface

> If I click on the key behind Home (the one with the house icon) I get the following answer:
G21 < ok
G53G90G0X287Y-178.3Z0 < ok
G92X287Y-178.3Z0 < ok
But nothing happen.

Button affects working coordinates, restores coordinates given by "Zero XY" & "Zero Z" buttons.

> I cannot find the G92 code in the list of codes supported by GRBL.

List of supported commands here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

> do not like the Comments very much.

Yes, should be fixed.

> Currently this option remain active with the last setting so you can face problems in the case of forgetting setting it back to 100%.

Be careful using this software.

> It would be nice to have the possibility to set the language manually.

Yes, would be nice.

Larepera, 2018-01-04:
Hi Denvi, thanks for your answers!

Yes, I found the G92 listed but it seems that the command did not work. I will do more tests.

Have a very happy and prosperous new year!

failureM, 2020-05-18:
[Sorry, but I need to use a Google translator] I recently bought a CNC 3018 pro with Candle 1.1.7 included. Unfortunately, I can not use this milling machine because I am not able to determine the position 0.0 for the X, Y and Z axes. I have a question: does any of you also have such a problem that after pressing the buttons setting the initial position and even if the spindle is high and then after pressing "home" the cutter always wants to stick to the ground? If you know the way, please help.

closed 2025-09-01 by github-actions[bot]