Denvi/Candle #189: Приложение запускается и тут же завершает работу

ZloAlien, 2017-12-28:
Приложение запускается и тут-же завершает работу. Сообщений об ошибках не выводит. Ошибка повторяется на 3 компьютерах (один из них с Linux). На одном компьютере (Windows 10 x64) приложение удалось запустить с N-ной попытки, запуская его через консоль. После этого даже стандартный запуск происходит нормально. Экспериментально установлено что за "незапуск" приложения ответственна строка "formGeometry..." в файле settings.ini. Аналогичное поведение имело место даже при перекомпиляции через в QTCreator. 
[settings-не-работает.txt](https://github.com/Denvi/Candle/files/1591009/settings-.-.txt)
[settings-работает.txt](https://github.com/Denvi/Candle/files/1591010/settings-.txt)

Denvi, 2017-12-28:
В #173 подобная проблема была. У вас два монитора?

ZloAlien, 2017-12-29:
На 2 компьютерах из 3 проблема действительно была в двух мониторах. А на последнем похоже что нет. На нем установлен Windows 7 x32.

ZloAlien, 2017-12-29:
В журналах Windows имеется следующий отчет:
Имя сбойного приложения: Candle.exe, версия: 1.1.7.0, отметка времени: 0x03916f80
Имя сбойного модуля: ig4icd32.dll, версия: 8.14.10.1930, отметка времени 0x4aba6fc2
Код исключения: 0xc0000005
Смещение ошибки: 0x002095ed
Идентификатор сбойного процесса: 0xbbc
Время запуска сбойного приложения: 0x01d38083518ee73d
Путь сбойного приложения: F:\Candle\Candle.exe
Путь сбойного модуля: C:\Windows\system32\ig4icd32.dll
Код отчета: 8f710154-ec76-11e7-8cb8-001fe25bcae5

Denvi, 2017-12-29:
Посмотрите в сторону дайверов видеокарты, поддерживаемую версию OpenGL.