Denvi/Candle #191: Управление по bluetooth

halny123, 2017-12-30:
Здравствуйте.
Возможно ли сделать управление не только по СОМ но и через bluetooth ? Что бы не тянуть провода к станку ?

Denvi, 2017-12-30:
Приветствую.
Посмотрите модули HC-05.

ASHIT-AXAR, 2017-12-31:
[image]

[image]

and set HC_0x as auto connect.

artchizhov, 2018-01-03:
Уже пользуюсь давно, использую модуль HC-06, бывают иногда косяки в виде фризов, но у меня в ноуте какой то косяк с радиомодулем, так что может быть эти мелкие проблемы у вас вообще будут отсутствовать.

ASHIT-AXAR, 2018-01-06:
да есть некоторые проблемы, связанные с этим.
например, когда cnc выключен и подключен Bluetooth-ключ. если я открою Свечу, она фриз.
а также некоторые шумы во время поворота шпинделя при приостановке состояния, потому что я не использую никаких щитов с ардуино. все они невежественны.
иногда они разъединяются, прежде чем я нажимаю кнопку отправки, когда я устанавливаю задание.
со всеми этими проблемами, во всяком случае, я предпочитаю Bluetooth по кабелю.

closed 2025-09-01 by github-actions[bot]