Denvi/Candle #192: probe two times

snapedumbledore, 2017-12-30:
Hi, Denvi, the Candle is great.
I would like to know how can you realize the function that when you press 'probe', it will probe twice a time.
I can't fine the code about the function, Can you help me?
Thanks a lot.

Denvi, 2017-12-30:
Hello.
Service > Settings > Control > Probe commands

By default: 
~~~
G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10
~~~

snapedumbledore, 2017-12-30:
I know that in Candle.
I would like to know how can you realize it in source code .

snapedumbledore, 2017-12-30:
I can't fine the 'probe' source code.
[image]

Denvi, 2017-12-30:
In "frmMain.cpp"

~~~
void frmMain::on_cmdTouch_clicked()
{
//    m_homing = true;

    QStringList list = m_settings->touchCommand().split(";");

    foreach (QString cmd, list) {
        sendCommand(cmd.trimmed(), -1, m_settings->showUICommands());
    }
}
~~~

snapedumbledore, 2017-12-30:
Thank you very much.
I hope i can understand it.

efital1, 2018-01-05:
False probe reading - I added .5uF to A5 and GND. All false probe readings are gone.

mikypolice, 2018-12-07:
> False probe reading - I added .5uF to A5 and GND. All false probe readings are gone.
 .5uF ??
Electrolytic capacitor? Ceramic? 0.5uF isn't a commercial capacity... 0.47uF or 470nF is a commercial capacity... Wich you used specifically? size anda type... Thank's...

closed 2025-09-01 by github-actions[bot]