Denvi/Candle #199: Slow visualization on Win7 with AMD drivers but good with standard VGA drivers

TheMasterFX, 2018-01-05:
Hi,
because my CNC has grbl 0.9 I installed Candle 1.0.11.
But the visualization performance is very bad (~20fps with not gcode loaded and 1 fps with gcode displayed. I get the same with older and newer versions of Candle too). And one CPU is utilized 100% by candle and crashes when milling very often.
I'm using and old dual core (Core2Duo T5200) notebook with AMD Mobility Radeon x1350 running on Win7 64 bit.
Now comes the interesting part. When I use the "Standard VGA Drivers" the performance is very good: 60 fps with gcode loaded and displayed!!

So I wondered if Candle falls back in an faster OpenGL mode when no "native driver" is installed. But it might also be that the AMD drivers are broken. Unfortunately these are the latest and only drivers which are out there (supplied by Microsoft http://www.catalog.update.microsoft.com/Search.aspx?q=x1350)

Normally I would be fine using standard VGA drivers, but with them I'm not able to use the native resolution of my 16:10 Display since standard VGA only supports 4:3 resolutions.

So maybe there is a way to to force Candle to use the "fast" mode which is used with standard VGA drivers.

christoph1703, 2018-03-16:
Hi,
I have the same problem, although on a different system. I'm using Grbl 1.1f and Candle 1.1.7 on Linux Mint with an i7 CPU and GTX750 GPU.
I see one CPU core at 100% and rather bad performance rendering and sending GCode. Trying to jog continuously makes the machine move in little hops instead of a continuous movement, as Candle can't send the jog commands fast enough.
Interestingly, I don't have any problems with a freshly unzipped version of the program, so it might have something to do with settings. I have only set a few custom commands for probing and save position, so it might be some automatic setting which is hidden from view?

JDCodeIt, 2019-05-03:
I compiled Candle on a Pi 3B, and get the same very slow performance. 1.7 of the 4 cores are used when rendering and it takes a good 10 seconds to drop menus and click buttons.

closed 2025-09-01 by github-actions[bot]