Denvi/Candle #201: Heightmap not editable

Miq1, 2018-01-07:
I am using Candle 1.1.7 on a Win10 64bit PC. Milling etc. works, but I am not able to modify the height map at all. The "Heightmap Settings" area in the Candle window is all greyed out, showing the default settings. I was able to at least modify the size of the height map to fit my PCB with a test editor in the map file, but nothing else.

"Keyboard Control" in Jog section is checked of course.

What did I miss?

vdesmedt, 2018-01-13:
Hi, same config, same problem, same basic checks here.  Any news on this ?

Nandox7, 2018-01-19:
I was about the report the same.
Is somehow related to the "Keyboard Control" option, if you disable it the heightmap option get enabled.

At the moment I disable it every time I want to create a new heightmap but ideally this wouldn't be required.

docteurzoidberg, 2018-01-25:
uncheck keyboard control for now

(hello bender)

nickm324, 2020-08-12:
Keyboard option is not enabled and have the same issue. Any fix for this yet after 2 years?

ramiabraham, 2021-11-06:
Same issue here; have tried the above-noted checks as well.

justincdotme, 2021-12-10:
Does the label "wontfix" just mean that height mapping is no longer available in Candle?

Karquinyoli, 2022-01-21:
Just open settings, Panels, check HeightMap.

HourGlss, 2022-01-28:
This is still a problem for me. The heightmap shows but "Use Heightmap" checkbox is disabled.

Karquinyoli, 2022-01-28:
> This is still a problem for me. The heightmap shows but "Use Heightmap" checkbox is disabled.

I proceed as follows:

1.- Load nc file
2.- Load or Create  map in the HeightMap panel
3.- Check 'Use heightmap'

closed 2025-09-01 by github-actions[bot]