Denvi/Candle #209: Version 1.1.7 jog not working

andired, 2018-01-24:
Hello Denvi,

It seems I have the same issue with Candle 1.1.7 as well. Jog doesn't work at all.
Tested with other software and the hardware works ok.
Below are my settings

[CTRL+X] < Grbl 1.1f ['$' for help]
$ < [HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok
$$ < $0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
ok

docteurzoidberg, 2018-01-25:
hi, in 1.17, i was able to use all keyboard jog function, except the step change ones*

andired, 2018-01-25:
it doesn't work for me in candle but it works in grblcontroller, however in grblcontroller since I'm using grbl 1.1f I'm having all the time "status unknown"

ASHIT-AXAR, 2018-01-25:
@andired 
you need to upload the latest GRBL software into your arduino.

andired, 2018-01-25:
@ashit-axar if you read my initial post you can clearly see I have the latest grbl version 1.1f

ASHIT-AXAR, 2018-01-25:
@andired 
i have the latest GRBL and candle. it works well for me.
something is not correct with your settings.
try removing the current GRBL library completely by deleting its folder (For information on installing libraries, see: http://arduino.cc/en/Guide/Libraries)
and upload a fresh one.

andired, 2018-01-25:
it works just fine with another software similar with candle. the issue itself is with candle

Kim9999, 2018-02-09:
Yes I have this problem too.  Cannot get motors to run in jog Shows they are but nothing moves.  All updated To grbl 1.1f and using candle 1.1.70.  My eyes are hurting  please a simple answer? What piece of code do I need to add?  Thank you

Kim9999, 2018-02-11:
#213 e end it is an EASY fix.   Just add this to you

Kim9999, 2018-02-11:
#

andired, 2018-02-11:
Kim9999 was this the solution?

$J=G21G91X0.7F115000
$J=G21G91Y0.7F115000
$J=G21G91Z-0.7F112000

Kim9999, 2018-02-11:
Yes  I am sorry still learning how to do forum.  Yes this did solve the
proble.   Some tweeks here and ther and all is working.   Thank you for
your forum helped out a lot.

andired, 2018-02-11:
it doesn't resolve the issue for me. what did you do exactly to solve this?

Kim9999, 2018-02-12:
Ok here is what I did was to enter the code and I did too think it was doing nothing.  After looking at my machine I found it was going very slow in jog.  turned it up and all was well..   Blew the program out yesterday and had to reload.   Blew a dll.   I reloaded and nothing worked.  Did what I did above and it worked. I am using Grbl1.1f with candle 1.1.70. At a 10 feed you can barely see it move at 2000 it moves normal.   So this is where I am at now and it is cutting.  However I do not like the min feed of 10 I want the machine to go slower for metal cutting.  However the thing loaded the german version drat so I will most likly have to do this all over again.  lol
[CTRL+X] < Grbl 1.1f ['$' for help]
$$ < $0=10
$1=25
$2=7
$3=7
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=300.000
$131=180.000
acceleration=100
toolAngle=15
toolType=1
fps=60
queryStateTime=40
autoScroll=false
header=@ByteArray(\0\0\0\xff\0\0\0\0\0\0\0\x1\0\0\0\x1\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\x6\x30\0\0\0\x2\0\0\0\x5\0\0\0}\0\0\0\x4\0\0\0}\0\0\x3\xf7\0\0\0\x6\0\x1\0\0\0\0\0\0\x1\0\0\0\0\0\0\0\x64\0\0\0\x32\0\0\0\x84\0\0\0\0\0\0\0\x6\0\0\0\x63\0\0\0\x1\0\0\0\0\0\0\x1\x5\0\0\0\x1\0\0\0\0\0\0\0\x9b\0\0\0\x1\0\0\0\0\0\0\x1\xf4\0\0\0\x1\0\0\0\x1\0\0\0\0\0\0\0\x1\0\0\0\0\0\0\0\0\0\0\0\x1\0\0\0\0\0\0\x3\xe8\0)
splitter=@ByteArray(\0\0\0\xff\0\0\0\x1\0\0\0\x2\0\0\x1\xd0\0\0\x1\x39\x1\0\0\0\f\x1\0\0\0\x2\0)
formGeometry=@ByteArray(\x1\xd9\xd0\xcb\0\x2\0\0\0\0\x4\xf8\xff\xff\xff\xf8\0\0\n\a\0\0\x4\a\0\0\x5\xa1\0\0\0Z\0\0\tX\0\0\x3[\0\0\0\x1\x2\0\0\0\x5\0)
formSettingsSize=@Size(751 599)
feedOverride=true
feed=100
heightmapPanel=false
spindlePanel=true
feedPanel=true
jogPanel=true
keyboardControl=false
autoCompletion=true
units=0
storedX=-372.96600000000001
storedY=-5.6660000000000004
storedZ=0
recentFiles=C:/Users/Kim/Documents/gcode examples/GRBL.DXF.V2/GRBL V2.dxf, C:/Users/Kim/Documents/gcode examples/smile.nc, C:/Users/Kim/Documents/gcode examples/Alien Face.nc
recentHeightmaps=@Invalid()
touchCommand="G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10"
safePositionCommand="G21G90; G53G0Z0"
panelHeightmapVisible=true
panelSpindleVisible=true
panelFeedVisible=true
panelJogVisible=true
fontSize=9
consoleMinHeight=147
heightmapBorderX=-0.25
heightmapBorderY=0
heightmapBorderWidth=13.25
heightmapBorderHeight=19.620000000000001
heightmapBorderShow=true
heightmapGridX=2
heightmapGridY=2
heightmapGridZTop=1
heightmapGridZBottom=-1
heightmapGridShow=true
heightmapInterpolationStepX=20
heightmapInterpolationStepY=20
heightmapInterpolationType=0
heightmapInterpolationShow=true
ToolpathZMovement=#ff0000
ToolpathHighlight=#9182e6
ToolpathStart=#ff0000
ToolpathEnd=#00ff00
VisualizerBackground=#ffffff
VisualizerText=#000000
Tool=#ff9900
ToolpathDrawn=#d9d9d9
ToolpathNormal=#000000
grayscaleSegments=false
grayscaleSCode=true
drawModeVectors=true
laserPowerMin=0
laserPowerMax=255
lastFolder=C:/Users/Kim/Documents/gcode examples

recentCommands="$132=30", "$131=180", "$130=300", "$3=7", $$, "$3=4", "$5=Z", "$5=Z00000101", 
"$3=Z", "$J=G21G91X0.7F9000000\n$J=G21G91Y0.7F9000000\n$J=G21G91+Z0.7F2000000"

userCommandsPanel=true
panelUserCommandsVisible=true
userCommands1=G1
userCommands3=G3
userCommands2=G2
userCommands0=G0
ignoreErrors=true
autoLine=false
panelOverridingVisible=true
feedOverrideValue=1
rapidOverride=true
rapidOverrideValue=100
spindleOverride=true
spindleOverrideValue=95
jogFeeds=5, 50, 100, 500, 1000, 2000
jogFeed=2000
jogSteps=Continuously, 0.01, 0.1, 1, 5, 10, 100

I set out the entry above so you could see what I put into the settings on the comand line.     This worked and is working now.   Hope this helps you out.

lexelby, 2019-04-20:
In my case, I had to enter a value into the "feed" box in the Jog panel before it would actually do anything.

smeeminator, 2021-06-09:
similar for me except there was a figure in 'feed box' but it was 10, changed it to 500, boom, happy days

magghub, 2022-07-21:
In my case is not working while use $J= at line begining, if I erase the "$J=" works fine.

[CTRL+X] < Grbl 0.9j ['$' for help]
$J=G21G91X0.000Y-5.000Z0.000F10 < error: Bad number format
$J=G21G91X0.000Y5.000Z0.000F500 < error: Bad number format
G21G91X0.000Y5.000Z0.000F10 < ok

SzymCode, 2024-07-24:
Nothing helped for me, I came up with the idea to click the "Reset" button on the motherboard and BOOM! 
Everything works fine :)

closed 2025-09-01 by github-actions[bot]