Denvi/Candle #210: How I create GCODE code for Candle?

StefanosBand, 2018-01-25:
What kind of software I have to use in order to create GCODE code for the Candle application? 
For example I want to convert an image in GCODE for Candle.
Thanks in advance and best regards to everybody!!!!!

ASHIT-AXAR, 2018-01-26:
@StefanosBand 
you need a "CAD" software to do your design and a "CAM" software to generate CNC GCODE from a CAD model.
there are various softwares. some of them combine those in a one package.
i personally use "Vectric Aspire".
"ArtCam" is the most famous software.

StefanosBand, 2018-01-26:
Dear Friend ASHIT-AXAR

 

I will look for the software that you suggest to me.

Do you know if this software can create a compatible GCODE for the Candle program?

Because some of the software that I have try, send wrong or incompatible commands to the CNC machine ( mine is the CNC 3018 mini CNC router/engraver).

Thanks a lot again and best regards !!!

Stefanos
Konstantopoulos
GREECE-CRETE

 
 

Sent: Friday, January 26, 2018 at 8:19 AM

ASHIT-AXAR, 2018-01-26:
Candle is a control software and sends your gcode file to GRBL device through serial port.
thus Candle stands out of the gcode program and it's content.
you need to find the GRBL gcode file format in the CAM output.
for example:
[image]
i'm using the highlighted item (gcode format) for my grbl.

StefanosBand, 2018-01-26:
Ok I understand. I 'll try the same to see what I get.

Thank you very very much for your help!!!!

 

Stefanos
Konstantopoulos
GREECE-CRETE

 
 

Sent: Friday, January 26, 2018 at 11:49 PM

StefanosBand, 2018-02-15:
Hello Dear ASHIT-AXAR

 

First of all I want to thank you very much for your Help!!!

I tried the Aspire and everything worked out fine except the scaling.

I create images in Corel Draw and I output the file as *.svg save it in pixels in 1:1000 accuracy but when I upload it in Aspire I do not have the same scale in the routing result as it is inside the Aspire program.

Example for a square 100x100mm :

 

CorelDraw:                             square 100x100mm

Aspire:                                   square 100x100mm

Routing result (CNC3018):       10x10mm approximately

 

I have found that 720mm in Aspire=180mm on the routing result.

What I do wrong???

The routing machine is the Ebay CNC3018 mini routing machine.

 

Many Greetings and Have a nice day!!!

Stefanos
Konstantopoulos
GREECE-CRETE

 
 

Sent: Friday, January 26, 2018 at 8:19 AM

ASHIT-AXAR, 2018-03-08:
@StefanosBand 
oh, its about one month i didnt check my email (my only email) :sweat_smile: . so, i dont know you find the solution already.
try saving your design as *.EPS or *.DXF
if both settings are in mm then everything should be fine.

StefanosBand, 2018-03-08:
Hello Dear Ashit Axar

 

Ok I'll try this.

Thank you very very much!!!!

 

Stefanos
Konstantopoulos
GREECE-CRETE

 
 

Sent: Thursday, March 08, 2018 at 7:36 PM

DEF7, 2018-10-07:
My program PixelCNC is great for generating a variety of toolpaths from simple 2D image input, if you don't want the fuss of 3D CAD/modeling. It's still being actively developed, and is available for early-access as a public alpha version, but I've been releasing updates for it regularly since its initial release in January.

StefanosBand, 2018-10-07:
Thank you all for your Help!!!!

closed 2018-10-07 by StefanosBand

KARPOVS, 2018-10-14:
> What kind of software I have to use in order to create GCODE code for the Candle application?
> For example I want to convert an image in GCODE for Candle.
> Thanks in advance and best regards to everybody!!!!!

Hello Stefanos, 

I also bought a 3018 cnc. I'm using the spire, but I'm having problems generating the G-CODE, the CNC 3018 is not interpreting the commands correctly. What kind of G-CODE output format are you saving on ASPIRE?
TKS

StefanosBand, 2018-10-14:
I saved as	*. tap file type. But I have problems with the scale. It does not have the same size in the drawing and the real object!!!
Have a good day and best regards😊👍🌟. 
-- 
Sent from my Android phone with mail.com Mail. Please excuse my brevity.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Denvi/Candle/issues/210#issuecomment-429627165",
"url": "https://github.com/Denvi/Candle/issues/210#issuecomment-429627165",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [Denvi/Candle] How I create GCODE code for Candle? (#210)",
"sections": [
{
"text": "",
"activityTitle": "**KARPOVS**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@KARPOVS",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Denvi/Candle\",\n\"issueId\": 210,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Denvi/Candle/issues/210#issuecomment-429627165"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 297139188\n}"
}
],
"themeColor": "26292E"
}
]

KARPOVS, 2018-10-14:
I use SPIRE V8, when I record G-CODE there are several formats with * .tap, such as "G-CODE ARCS", "CANMASTER", "MICROMIL" ... etc, which one do you use?
I am using to control CNC 3018 software CANDLE, do you use this or other?
TKS

StefanosBand, 2018-10-14:
Hi again

 

I use the  >> G-CODE ARCS(mm).tap type of file.

The told to me to use this type o file.

I amm not an expert in this CNC software.

this person knows better >>> ASHIT-AXAR <notifications@github.com>

 

Send him your question and maybe will help you better.

 

Bye bye for now and best regards !!!!

 

 

 

Stefanos
Konstantopoulos
GREECE-CRETE

 
 

Sent: Sunday, October 14, 2018 at 5:17 PM

ASHIT-AXAR, 2018-10-17:
@StefanosBand is your problem with scaling still not fixed ?
you can message me in Telegram. ( @ashit_axar )

StefanosBand, 2018-10-17:
Hello Dear ASHIT AXAR Again

 

Yes I still have problem with the scaling!!

I use Aspire 8.5 and I save files as G-CODE ARCS(mm).tap (as you told me then)

I draw files in Corel Draw and I save them as .svg, then I load it in Aspire and then I save it as G-CODE ARCS(mm).tap

Actually I have discovered that:

(1mm) in Corel draw equals to 4mm in Aspire.

What I do wrong ???

Thanks for talikng again to you and thanks for your help too!!!!

Have a good day and best regards !!!!!!!!

 

Stefanos
Konstantopoulos
GREECE-CRETE

 
 

Sent: Wednesday, October 17, 2018 at 3:18 PM

ASHIT-AXAR, 2018-10-18:
@StefanosBand this make no sense :thinking: 
let's test your machine at first place. run this file on your CNC and report the result.

[test.zip](https://github.com/Denvi/Candle/files/2493169/test.zip)

code sequence:
1. prepare environment.
2. go to z10 ( 10mm above zero point of working position).
3. go to z0.
4. go to 100mm in x+ direction

it moves head in x+ direction 100mm. (measure this distance, it should be 10cm)

and also let's see your CNC config. ( type $$ command in Candle console).

StefanosBand, 2018-10-18:
Ok I'll try that and I let you know the results.

Thanks a lot for your Help!!!!!!

 

 
 

Sent: Thursday, October 18, 2018 at 9:35 PM

let's test your machine first. run this file on your CNC and report the result.

	prepare environment.
	go to z0.

StefanosBand, 2018-10-19:
$0=10 (step pulse, usec); 
$1=25 (step idle delay, msec); 
$2=0 (step port invert mask:00000000); 
$3=5 (dir port invert mask:00000101); 
$4=0 (step enable invert, bool); 
$5=0 (limit pins invert, bool); 
$6=0 (probe pin invert, bool); 
$10=3 (status report mask:00000011); 
$11=0.010 (junction deviation, mm); 
$12=0.002 (arc tolerance, mm); 
$13=0 (report inches, bool); 
$20=0 (soft limits, bool); 
$21=0 (hard limits, bool); 
$22=0 (homing cycle, bool); 
$23=0 (homing dir invert mask:00000000); 
$24=25.000 (homing feed, mm/min); 
$25=500.000 (homing seek, mm/min); 
$26=250 (homing debounce, msec); 
$27=1.000 (homing pull-off, mm); 
$100=200.000 (x, step/mm); 
$101=200.000 (y, step/mm); 
$102=200.000 (z, step/mm); 
$110=5000.000 (x max rate, mm/min); 
$111=5000.000 (y max rate, mm/min); 
$112=500.000 (z max rate, mm/min); 
$120=20.000 (x accel, mm/sec^2); 
$121=20.000 (y accel, mm/sec^2); 
$122=20.000 (z accel, mm/sec^2); 
$130=200.000 (x max travel, mm); 
$131=200.000 (y max travel, mm); 
$132=200.000 (z max travel, mm); 
ok

ASHIT-AXAR, 2018-10-19:
do you have datasheet of your cnc ? (if so, upload here to check for settings)
your settings may not be correct for your grbl.
i think $100, $101, $102 is too low. (maybe some values around 800 are more reasonable) 
also, if max travels are not set correctly, may damage your machine.

StefanosBand, 2018-10-19:
Hi Ashit

 

I have not any datasheet but I found this link in the web that has all the information you need for the CNC3018 that I have too.

 

http://ejs.seniejitrakai.net/2018/01/20/my-grbl-settings-for-3018-cnc-diy-machine/

 

Before I put the machine in operation I double check the maximum values and afterwards I send the file to the machine.

 

What about what I told you for the measurments 100mm is 10 cm.

Is this correct???

BUT in my michine 100mm is 25mm 2,5 cm.

 

Thanks a lot Stefanos.

 

 

 

Stefanos
Konstantopoulos
GREECE-CRETE

 
 

Sent: Friday, October 19, 2018 at 1:16 PM

ASHIT-AXAR, 2018-10-19:
> BUT in my michine 100mm is 25mm 2,5 cm.
ok, i think this is the problem. 
set these values on by one in the Candle console:
$100=800
$101=800
$102=800

this should fix the issue

StefanosBand, 2018-10-19:
Ok I will change the values, but I don't know how to do that!
Please tell me how I do this.
Thanks in advance and have a great day
😊😊😊👍👍👍
-- 
Sent from my Android phone with mail.com Mail. Please excuse my brevity.

"activityTitle": "**ASHIT-AXAR**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@ASHIT-AXAR",

ASHIT-AXAR, 2018-10-19:
turn on the cnc. open the candle and when the connection is ok, type these lines into console one by one and hit send button:

$100=800

$101=800

$102=800

you should get an 'ok' message for each.

StefanosBand, 2018-10-20:
OK I got it!!! 🌟 👍 👍 😊 Many many thanks for your help!!! I will do it as soon as I get back home. Next week. I will let you know when I have results!!! 🌟 👍 👍 😊 😊 😊 👍 👍 👍