Denvi/Candle #212: Compilation fails under Ubuntu 16.04 amd64

DmitryPan, 2018-01-30:
Hello and thanks for great app!

Unfortunately compilation fails under Ubuntu 16.04 amd64, Qt 5.5.1. I have already installed dependency packages mentioned in previous posts and done the following steps.
```
git clone https://github.com/Denvi/Candle.git
cd Candle/src
qmake -project
qmake
make
```

The output of the last command is
```
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -I. -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o frmabout.o frmabout.cpp
In file included from frmabout.cpp:5:0:
frmabout.h:7:19: fatal error: QDialog: No such file or directory
compilation terminated.
Makefile:606: recipe for target 'frmabout.o' failed
make: *** [frmabout.o] Error 1
```
I have QDialog in /usr/include/x86_64-linux-gnu/qt5/QtWidgets directory, so I tried running manually
```
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -I. -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o frmabout.o frmabout.cpp
```
which outputs:
```
frmabout.cpp:6:25: fatal error: ui_frmabout.h: No such file or directory
```

Could someone please help me compiling this app? Maybe someone has pre-built DEB package?

Saur0o0n, 2018-02-12:
Try "qmake candle.pro; make"

unllama, 2018-03-25:
I had similar issues on an identical configuration; ultimately solved by remembering to be in /src , then just qmake and make (.pro already existed)

closed 2025-09-01 by github-actions[bot]