Denvi/Candle #216: Yet another visualization garbae (Linux/Intel)

Saur0o0n, 2018-02-12:
Hi,
I've compiled latest Candle on my Debian (sid). I have completely no visualization (it shows current video card buffer - for example, currently played movie). Candle requirements states there is need for OpenGL 2.0 - as far I can see it's there (Intel i7-640LM):

`sn@warsztat:~$ glxinfo | grep -i opengl
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Ironlake Mobile 
OpenGL version string: 2.1 Mesa 17.3.3
OpenGL shading language version string: 1.20
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 17.3.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
OpenGL ES profile extensions:
`
I've tried every possible visualization settings in Candle - no luck. Any advice?

WorkOfArtiz, 2018-02-20:
Having the exact same issue with an Intel Graphics 620, running OpenGL 3.0 on Ubuntu.

dunkelstern, 2018-02-20:
Same here: Intel HD Graphics 3000 512 MB (MacBook Pro 2011), seems to be related to the intel driver then. Using most recent version on arch linux. No change if running on wayland or x11

Saur0o0n, 2018-02-20:
It can't be just Intel driver - on another Debian of mine (i5-4690, Intel HD 4600) is a bit better. Black background, gcode not entirely visible - but with grey scale it's partially ok.

dunkelstern, 2018-02-20:
Something is broken with it on newer Qt versions it seems, does not display anything on macOS too. Using Qt 5.10 here.

Saur0o0n, 2018-02-20:
Mine is 5.9.2+dfsg-12 - on both hosts

dunkelstern, 2018-02-20:
Ok, settings were borked here, with Qt 5.7 it works but stutters, with 5.10 it gets a solid 60fps. Check if the visualizer settings are set to all colors fully black (they were here. It even seems that they were not set at all so the drawing didn't work at all)

Saur0o0n, 2018-02-20:
Ok - so it is ok then (colours problem) on i5-4690, Intel HD 4600. On Intel i7-640LM it does not change anything - garbage on the screen.

dunkelstern, 2018-02-20:
On macOS it works for me, on linux with the radeon card active it works too, no success with the intel though

yuranikspb, 2018-04-23:
windows qt 5.7.0 MiniGw   blackscreen (grayscale setting make nc code visible)

Saur0o0n, 2019-06-13:
Ugly workaround for this problem is to force software opengl rendering, by setting environment variabe
LIBGL_ALWAYS_SOFTWARE=1

YdalirU, 2022-09-27:
I just recently compiled Candle 1.1.8 under LINUX Mint 20, QT 5.12.8.
lots of stuff I do not know  whether it works or not (like klicking the HOME button does not HOME ..)

Anyway, the current issue is that I have no visualization, just (as stated in this thread) "Garbage" from the video buffer.

The GRBL File was build with INKSCAPE, but since GRBL is some standard G-Code like, I do not see an issue here.
My Laptop is 
Intel® Core™ i5 CPU M 520 @ 2.40GHz × 4
 4GB Memory,
Mesa DRI Intel® HD Graphics (ILK).
Version Linux Mint 20.3 Una 64-bit
Kernel Linux 5.4.0-125-generic x86_64
MATE 1.26.0

Any Idea what might prevent the visualizer to work? 

Generic Functionality is OK, the circle (as given in the nc file (renamed to .txt for upload) is milled.

[CNC 160mm Kreis_0004.txt](https://github.com/Denvi/Candle/files/9658010/CNC.160mm.Kreis_0004.txt)

Thx a lot

closed 2025-08-31 by github-actions[bot]