Denvi/Candle #218: Verify machine with JOG function

shabbee, 2018-02-23:
Can anyone please confirm which way the table should move when I jog my machine -Y? My machine moves in the opposite direction to the visual in the program, same applies with jogging X.

Z jogs up and down correctly

Thanks again for any advice

ASHIT-AXAR, 2018-02-27:
@shabbee 
-Y should move to negative (or to zero as your config) (towards you or down)
if it goes in wrong direction, you have two option:
1. change wiring order.
2. change grbl direction config (set in console)

Kim9999, 2018-04-11:
Here is the info you need. https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

shabbee, 2018-04-11:
Thanks guys, I eventually figured it out but it's good to have the reassurance that I found the correct solution

closed 2018-04-11 by shabbee