Denvi/Candle #221: Newly created height map is not automatically enabled

jasperoosthoek, 2018-03-11:
Hi,

I've been using Candle to create pcbs using isolation routing and the following situation has occured several times already. I probe a board and save the height map thinking that I've made a height map and everything is fine. Then after switching back to the g-code section and press send, the probed height map is not used. Also in the height map section the "Use heightmap" switch box is gray it is easy to overlook it.

So it requires an extra step to actually use the height map and there is no warning when you don't use it.

I propose a couple of possible solutions:
- Automatically turn the height map switch box on. Idea is that it doesn't make sense to probe a height map an then not using it.
- Give some sort of warning to the user when you press send and a height map is loaded but not used: "Height map is not used, continue anyway?"
- Removing the switch box altogether. The reasoning being, why probing a height map when you do not plan using it.

Thanks in advance, Jasper

DarrenSchwartz, 2020-04-19:
Oh man!  Thank you so much for this two year old post!  I was pulling my hair out after running the height map routine several times and still could not mill through a small bit of copper.  This check box should definitely be on by default.  As you say, if someone just went through the process of mapping it out, there is a near 100% chance that they want to use it.  Thank you again!

jasperoosthoek, 2020-04-20:
You're very welcome! I might be producing a couple more boards in the near future and will keep you updated if I notice more things.

Other improvement would be:
-  Ability to extrapolate outside of the height map instead of only interpolate in between points. Then you can have PVCs larger than the height map. This useful when you need to use as much space on the board as possible. Say you would like to use the full width of a 100mm board. It makes perfect sense to zero X and Y at a corner and probe from say X = 2mm to X = 98mm. Then you only have to cut off one side of the board.
- As boards are flexible I found out that sometimes the etching is much deeper than other times. It's hard to wrap my head around why but it could be the probe speed. Could you add the ability to change the probe speed?

I hope you might find the time to incorporate these ideas which would greatly improve my work flow!

ondras12345, 2020-04-20:
> Could you add the ability to change the probe speed?

Settings > Heightmap > Heightmap probing feed

jasperoosthoek, 2020-04-21:
Ok nice

closed 2025-08-31 by github-actions[bot]