Denvi/Candle #226: Using candle 1.1.7 and jog not working at all

Morphiritus, 2018-04-02:
using windows version 1.1.7 candle and no Jog. However unlike the Grblcontrol version i noticed that no codes are sent to console window. if i type a code machine moves. Appears there's a missing link between arrows and send area. Would love to work with candle and try it out but messy when i have to set head in grbl then close software and open candle. Is it possible to get new windows build to try. Thanks if can if not that ok too.  I'm new to this and am trying not to relearn linux . Been few years and mind not what it was.
I can't find grbl version 1.1 so unable to try that. Having fun so far though.

Kim9999, 2018-04-11:
Here is grbl 1.1f   https://github.com/gnea/grbl/releases  add this to get jog to work
$J=G21G91X0.7F200
$J=G21G91Y0.7F200
$J=G21G91Z-0.7F200
The value after f may need to be modified for your app.

Morphiritus, 2019-09-09:
Thank you for getting back to me. My brain isn't what it use to be and I
didn't realize my hobby CNC was only supported by your older version since
my CNC machine was an older version. I downloaded that version and all
worlds fine.

magghub, 2022-07-21:
> Here is grbl 1.1f https://github.com/gnea/grbl/releases add this to get jog to work $J=G21G91X0.7F200 $J=G21G91Y0.7F200 $J=G21G91Z-0.7F200 The value after f may need to be modified for your app.

The problem is the " $J= " erase it and CNC moves.  other version does not start comands with $J= and that why works. 
What I need for fix this issue?? somebody knows?

closed 2025-08-31 by github-actions[bot]