Denvi/Candle #227: Axes lock on start up alarm stet

Kim9999, 2018-04-11:
This just started out of the blue.   Will not run just locks axes.  Removed limit switch hook up still locks...Open software hooks up and locks up.

ASHIT-AXAR, 2018-04-12:
do not touch hardware. it's a safety feature in the software [GRBL] provided to locate precise machine zero.
i personally prefer to left it enabled. so during power loss or some kind of failure i can continue the job where i miss.

you can disable that in the Config.h file by commenting out line 88.
and upload this new firmware.

Kim9999, 2018-04-12:
Thank you.   I reset the eeprom and reloaded all settings so far it is
moving going to test it out today.

Kim9999, 2018-04-12:
Today I ran a file once no problem now all I get is error 9.  I am at a
mental block here sorry.

ASHIT-AXAR, 2018-04-13:
soft limits are a good safety feature. i left it enabled too. this assures machine will never damage itself.
you need to set correct values for all axes, so machine can work well.
related option's are : $130, $131, $132

Kim9999, 2018-04-13:
Yes I changed these to many settings and all locked it up.   I do have
limit switches and stop unit. I set the machine to size of bed and just has
a fit if I change $20.  It did work but now it does not after windows did
an update in the middle of a run. I turned off auto windows update!
Running air cuts now and all is working.  Will throw limit switches and
check all of that today.   Thank you very much!!!  I will send you my set
up when I copy the files to this computer.(lap top)
HAVE A WONDERFUL DAY!

Kim9999, 2018-04-15:
The lock up happened after a windows update during a run.  The only way I found to fix it was to reset the eeprom and reset everything. However I had to turn off soft limits as  it would throw an alarm code of 2 and an error code 9.  I set the soft limits to several setting with no improvement.  I set$20 to 0 now I am up and running. any thoughts on this?

ASHIT-AXAR, 2018-04-15:
first of all, post your grbl settings here.
grbl does not alter eeprom settings on its own. 
you do not need to change grbl source files by default
everything you need to config is listed in $$ settings.

Kim9999, 2018-04-21:
Ok here are my settings.  I use Vcrave Pro 6.9   I have to reverse X to get
text to print correctly and I have to invert x to get the letters A and V
to print correctly.  Just starting out so may have some settings wrong
still.  It is working but with the X going reverse of the buttons.  Left=
right.
Thank you for your help and advise.

$$ < $0=10
$1=25
$2=6
$3=6
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=1
$20=0
$21=1
$22=1
$23=0
$24=250.000
$25=500.000
$26=250
$27=8.000
$30=500
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=500.000
$111=500.000
$112=500.000
$120=2.000
$121=2.000
$122=2.000
$130=600.000
$131=360.000
$132=30.000
ok

$# < [G54:0.0000,0.0000,0.0000]
[G55:0.0000,0.0000,0.0000]
[G56:0.0000,0.0000,0.0000]
[G57:0.0000,0.0000,0.0000]
[G58:0.0000,0.0000,0.0000]
[G59:0.0000,0.0000,0.0000]
[G28:0.0000,0.0000,0.0000]
[G30:0.0000,0.0000,0.0000]
[G92:0.0000,0.0000,0.0000]
[TLO:0.0000]
[PRB:0.0000,0.0000,0.0000:0]
ok

$G < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0.0 S200]
ok

$I < [VER:1.1f.20170801:]
[OPT:V,15,128]
ok

$N < $N0=
$N1=
  I over set the 130,31,and 32 settings to get this to work.  I have a 300
mm x 180 mm bed or 7 x 12 inches.

Limit switches as shown in gnea/grbl in Wiki using NO switches in parallel
with the optiocoupler circuit there.   All that is working as it should.
With the woodpecker board.

Kim

Kim9999, 2018-04-21:
V crave Pro 6.0 not 6.9

ASHIT-AXAR, 2018-04-22:
ok, step by step to solve the problem.
here you inverted Y and Z. let's verify axes movements.
on the panel:
**for X**: pressing button right should move to the right "+X" "from negative to zero" or "from zero to positive", button left for opposite.
**for Y**: pressing button up should move the machine head away from you "+Y" (to zero or positive). button down for opposite.
**for Z**: pressing button up should move up "+Z" (to zero or positive). button down for opposite.

anything else is incorrect and you should invert direction mask ($3)

Kim9999, 2018-04-23:
Yes I have to invert X for text print correctly but I have to mirror the
text too. BUT it is not correct it is a work around. Reading about Machine
positions and work piece position  it has to be something I do not have
correct yet

Thank you
Kim Kottas

Kim9999, 2018-04-23:
I have tried several combinations to see what it would do and none work
unless I mirror my project and do work around on X.

ASHIT-AXAR, 2018-04-23:
ok, just ignore printing text or something (those are out of grbl hardware and it's settings)
are machine movements relative to the Candle control panel buttons ?
let's upload a picture of your machine.

closed 2025-08-31 by github-actions[bot]