Denvi/Candle #228: runtime error

houguang, 2018-04-12:
[image]
[image]
[image]
Error opening 50M file

houguang, 2018-04-12:
this is attachment 
[1.zip](https://github.com/Denvi/Candle/files/1903679/1.zip)

ASHIT-AXAR, 2018-04-13:
how much is your computer RAM size ?
you can split your file into smaller parts

houguang, 2018-04-13:
> how much is your computer RAM size ?
> you can split your file into smaller parts

8G

houguang, 2018-04-21:
this question no anwser?

closed 2018-04-21 by houguang