Denvi/Candle #233: Linux arm compiling error

yuranikspb, 2018-05-31:
I have cubieboard with A10 proccessor and try to compile candle
my errors.:

root@Cubian:/home/vnc/Candle/src# make
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic frmmain.ui -o ui_frmmain.h
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic frmsettings.ui -o ui_frmsettings.h
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic frmabout.ui -o ui_frmabout.h
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic widgets/sliderbox.ui -o ui_sliderbox.h
g++ -c -pipe -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIE -DUNIX -DGLES -DsNan="65536" -DAPP_VERSION=\"1.1.8\" -D_USE_MATH_DEFINES -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtOpenGL -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -o main.o main.cpp
In file included from main.cpp:18:0:
frmmain.h:72:24: error: expected \u2018;\u2019 at end of member declaration
frmmain.h:72:30: error: \u2018_GLIBCXX_USE_NOEXCEPT\u2019 does not name a type
frmmain.h:72:17: error: looser throw specifier for \u2018virtual const char* CancelException::what() const\u2019
/usr/include/c++/4.6/exception:69:25: error:   overriding \u2018virtual const char* std::exception::what() const throw ()\u2019
frmmain.h:206:30: sorry, unimplemented: non-static data member initializers
frmmain.h:244:26: sorry, unimplemented: non-static data member initializers
frmmain.h:244:26: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_fileChanged\u2019
frmmain.h:245:31: sorry, unimplemented: non-static data member initializers
frmmain.h:245:31: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_heightMapChanged\u2019
frmmain.h:269:24: sorry, unimplemented: non-static data member initializers
frmmain.h:269:24: error: \u2018constexpr\u2019 needed for in-class initialization of static data member \u2018m_storedX\u2019 of non-integral type
frmmain.h:270:24: sorry, unimplemented: non-static data member initializers
frmmain.h:270:24: error: \u2018constexpr\u2019 needed for in-class initialization of static data member \u2018m_storedY\u2019 of non-integral type
frmmain.h:271:24: sorry, unimplemented: non-static data member initializers
frmmain.h:271:24: error: \u2018constexpr\u2019 needed for in-class initialization of static data member \u2018m_storedZ\u2019 of non-integral type
frmmain.h:280:28: sorry, unimplemented: non-static data member initializers
frmmain.h:280:28: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_settingZeroXY\u2019
frmmain.h:281:27: sorry, unimplemented: non-static data member initializers
frmmain.h:281:27: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_settingZeroZ\u2019
frmmain.h:282:21: sorry, unimplemented: non-static data member initializers
frmmain.h:282:21: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_homing\u2019
frmmain.h:283:33: sorry, unimplemented: non-static data member initializers
frmmain.h:283:33: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_updateSpindleSpeed\u2019
frmmain.h:284:33: sorry, unimplemented: non-static data member initializers
frmmain.h:284:33: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_updateParserStatus\u2019
frmmain.h:285:25: sorry, unimplemented: non-static data member initializers
frmmain.h:285:25: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_updateFeed\u2019
frmmain.h:287:23: sorry, unimplemented: non-static data member initializers
frmmain.h:287:23: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_reseting\u2019
frmmain.h:288:29: sorry, unimplemented: non-static data member initializers
frmmain.h:288:29: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_resetCompleted\u2019
frmmain.h:289:23: sorry, unimplemented: non-static data member initializers
frmmain.h:289:23: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_aborting\u2019
frmmain.h:290:29: sorry, unimplemented: non-static data member initializers
frmmain.h:290:29: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_statusReceived\u2019
frmmain.h:292:29: sorry, unimplemented: non-static data member initializers
frmmain.h:292:29: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_processingFile\u2019
frmmain.h:293:32: sorry, unimplemented: non-static data member initializers
frmmain.h:293:32: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_transferCompleted\u2019
frmmain.h:294:26: sorry, unimplemented: non-static data member initializers
frmmain.h:294:26: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_fileEndSent\u2019
frmmain.h:310:25: sorry, unimplemented: non-static data member initializers
frmmain.h:310:25: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_keyPressed\u2019
frmmain.h:311:23: sorry, unimplemented: non-static data member initializers
frmmain.h:311:23: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_jogBlock\u2019
frmmain.h:316:24: sorry, unimplemented: non-static data member initializers
frmmain.h:316:24: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_spindleCW\u2019
frmmain.h:317:34: sorry, unimplemented: non-static data member initializers
frmmain.h:317:34: error: ISO C++ forbids in-class initialization of non-const static member \u2018m_spindleCommandSpeed\u2019
make: *** [main.o] Error 1

yuranikspb, 2018-05-31:
QMake version 3.0
Using Qt version 5.3.2 in /usr/lib/arm-linux-gnueabihf

gcc version 4.6.3 (Debian 4.6.3-14)

guillaumepicquet, 2018-08-15:
Try with a greater gcc/g++ version.
I think **gcc** >= **4.9** may compile
Or try **clang++**

closed 2025-08-31 by github-actions[bot]