Denvi/Candle #234: Software cannot be restarted once closed.

dansperfect, 2018-06-24:
So this is my problem. I open up grblControl or the latest version Candle and it works fine. I close the program out and go about doing other things not necessarily on the pc. So i come back later to try a project out and grblControl or Candle no longer will start up.  Windows acts like it wants to start but nothing ever happens. So I copy either program back over the files that are already there and the program restarts again. I just don't understand whats happening. Can someone please help me with this issue.

Denvi, 2018-06-24:
Hi, may be #173 will help you.

dansperfect, 2018-06-24:
Thank you so much that works perfect now. I will make sure to use my main screen for now on.

closed 2025-08-31 by github-actions[bot]