Denvi/Candle #236: Looks like a great addition to grblControl

jakeb1, 2018-07-19:
Hi, I have installed grblControl 0.9 on my windows 10 and have played around with my small cnc. I now have seen your candle add it looks great. My questions are how I get it into the grblControl panel. I have some c programing experience but need help getting candle compiling with the grblControl. When I click
on E candle in the src file Eagle starts but I don't know where to go from there. I have pcb-gcode-3.6 the 
ULP  which allows me  to get my board cut image which I use in grblControl and I start my gcode file and my board is being cut. Your program is just what I need to make my height map but I am at a loss at what I need to do. I have read that my file needs to be a cam file. Do I have to add a directory to Eagle to produce this height map? I am just beginning in this so could use some guidance. Once I start making my boards I'll get my donation in. Thank you very much, Len

Denvi, 2018-07-21:
Hello, sorry, but I did not quite understand your question.
In general, the process of milling a printed circuit board should proceed as follows:
1) generate gcode-program using pcb-gcode for Eagle (*.tap files as I know);
2) load gcode-program with Candle;
3) create in Candle heightmap and apply it to gcode-program;
4) send modified gcode-program to CNC to mill.

There is video with heightmap creation process I found on youtube: https://www.youtube.com/watch?v=Ug-LOISRSOc, but in Russian language.

jakeb1, 2018-08-05:
Thank you for your replay. I was able to get a height map made from the grblControl panel and it seems to be okay. I do have a problem with the pads on my components they seem to be all connected with no isolation between them. When I use the ULP pcb-gcode I get a isolated board in eagle with the pads isolated from each other but when I go to cut the board with my tap file and the grblControl the pads on ics and resistors look to be not isolated and seem to be connected with the copper. Its like the pads are not being recognized. Any help would be appreciated, Len

jakeb1, 2018-08-06:
Hi, things are getting better for me and have figured the height map out. Is there a way to start in the middle of the program? I find that I need to pause for putting in the drill bits. I would like to reset my back to home then skip to a line further down in gcode. Thanks for your time. Len