Denvi/Candle #241: CNC hung at third step: (Output format description: G-CODE for LinuxCNC)

twinclouds, 2018-08-03:
I am using CNC 3018 for SainSmart.  It works fine so far.  However, yester, the machine hung at third step: (Output format description: G-CODE for LinuxCNC) for two particular g-code scripts.  It has no error but just stop there.  When I shut off Candle, it said it was sending the g-code.  Is there anyway I can find what happened?  Other g-code scripts works just fine.
Appreciate any help.

twinclouds, 2018-08-04:
I used bCNC to generate g-codes by the same dxf files and fed to my 3018 CNC 3018 for a dry run.  Since I don't know how to set the feed speed and cutting depth using, bCNC, I cannot to a real run.  But at least, the spindle moved just fine.  So the problem is likely caused by dxf2gcode program.
I can up load the dxf file if anybody want to give it a test.

twinclouds, 2018-08-05:
I figured out the reason the hung-up.  I named the .dxf file name in the form of 2mm02x.dxf.  In the generated g-code file, there is a line with the words "Created from file: Z:/Test/2mm02x.dxf".  It appears 2mm02 is a code word in the g-code, or interpreted by grbl as something to hang the execution of the code.  Once i changed name to something else, say 2mm-02x, everything is fine.
Maybe someone with good knowledge about g-code and/or grbl can tell me what was exactly happening. :-).

closed 2018-08-05 by twinclouds

jds2100, 2019-10-06:
One thing I have found thus far with the dxf2gcode translation is that at the begining of a block/shape it leaves out a feed rate, see examples below;

dxf2gcode original:

`G0 X  43.922 Y 125.679

M3 M8

G0 Z   1.500 

// Missing feed rate! //

G1 Z  -0.050

F80

G1 X  36.980 Y 125.679

G1 X  36.241 Y 126.421

F10

G1 Z  -0.100

F80

G1 X  36.980 Y 125.679

G1 X  43.922 Y 125.679

F10

G1 Z  -0.150

F80

G1 X  36.980 Y 125.679

G1 X  36.241 Y 126.421

F10

G1 Z  -0.200

F80

G1 X  36.980 Y 125.679

G1 X  43.922 Y 125.679

F10

G1 Z   1.500

G0 Z   3.000 

M9 M5`

Corrected:
`G0 X  43.922 Y 125.679

M3 M8

G0 Z   1.500 

F10 // ! Feed rate added to prevent machine stall at this point for bCNC ! //

G1 Z  -0.050

F80

G1 X  36.980 Y 125.679

G1 X  36.241 Y 126.421

F10

G1 Z  -0.100

F80

G1 X  36.980 Y 125.679

G1 X  43.922 Y 125.679

F10

G1 Z  -0.150

F80

G1 X  36.980 Y 125.679

G1 X  36.241 Y 126.421

F10

G1 Z  -0.200

F80

G1 X  36.980 Y 125.679

G1 X  43.922 Y 125.679

F10

G1 Z   1.500

G0 Z   3.000 

M9 M5`