Denvi/Candle #243: Does it now work on raspberry pi 3b+?

TB1405, 2018-08-14:
Question is above

aidagamemnon, 2018-09-02:
I can't compile it for raspberry coz it needs GLES/gl.h file and raspbian only have GLES2/gl2.h and GLES3/gl3.h.  How to compile it???

alexeytm, 2018-11-21:
Так как все таки собрать на raspberry pi? Ни с GLES2 ни с GLES3 не получается!

1f7, 2019-02-19:
`sudo apt install libgles1-mesa-dev`

AlexeyKichev, 2019-02-21:
Спасибо

dremin1984, 2019-03-10:
You can complete the installation guide candles in raspberry PI3? pls

TheNetStriker, 2019-11-13:
I've managed to compile Candle for my Raspberry Pi 3, but I'm having some weird problems. I'm new to QT and I hope someone can help me figure out what could cause those problems. (My knowlege about QT so far is from different issues here and from different guides I found) 

To compile the project I followed this guide: https://www.kampis-elektroecke.de/raspberry-pi/qt/?unapproved=10934&moderation-hash=0be1b2953f660c5853d0397cddf89ac3#comment-10934

I've used QT 5.10.1 to compile the project exactly as described in the gude. I also added the two lines that jvarl mentioned in [this ](https://github.com/Denvi/Candle/issues/77) issue and I was able to compile the project under Ubuntu and start it remotely on the Raspberry. I was also able to connect to an grbl Arduino, load some gcode and display the path in the 3d window.

The first problem I found is that the programm runs in some sort of Fullscreen mode and the Raspberry Desktop does not seem to be aware of this window. If I click somewhere in the application in the background the Raspberry desktop still handles the mouse clicks as if the windows is not there at all. I'm also not able to take screenshots or videos from the application because of this. In the screenshots and videos I'm just seeing the desktop and how the clicks are handled there.

The second problem I found is that when I start sending the gcodes to the Arduino the 3d view of the path gets completely destroyed. I can only see some scattered lines. But the spindle head is rendered correctly. Sadly I'm not able to take screenshots of this because of the first problem.

Also the application does not run at all on a Raspberry Pi 4. (There is just no window appearing, but no errors in the console)

Does anyone have an idea what causes this problems. Maybe @jvarl could help here?

pihnat, 2020-04-28:
[https://github.com/Denvi/Candle/issues/77](url)
More info here:  https://github.com/pihnat/Candle
Read the wiki on that site.

closed 2025-08-31 by github-actions[bot]