Denvi/Candle #244: Карта высот

Dem9ih4ik, 2018-08-21:
Доброго времени суток.
Подскажите пожалуйста в чм может быть проблема:
Делаю карту высот для фрезеровки печатной платы со следующими настройками:
размер платы - 62х101 мм
размер области снятия карты - 67х106мм
количество точек снятия - 22х54
сетка интерполяции - 400х400
Применяю снятую карту к файлу фрезеровки, но после цикла обработки местами чётко видно что медь была еле задета или не задета вообще.
На самом станке критических биений или пропусков шагов не наблюдается.
Что я мог не так сделать в настройках? И есть ли где-то документация о том каким образом вообще работой с картой, как она накладывается на g-code ?

pacmancoder, 2019-05-19:
Same problem here as @Dem9ih4ik described. 80x80 height map scan area, 11x12 scan grid produces strange results after applying it. While engraving, my z-axis suddenly lifted up too high, something around ~0.2mm, so most of the PCB was not engraved at all...  But what could be wrong with it?

I uploaded my gcode and map files here:

https://gist.github.com/pacmancoder/31a757fa21d35aa812eae34a5b479476

yuranikspb, 2019-07-24:
Таже проблема, при большом количестве замеров (  и сетки интерполяции) ось z заваливается вниз

ondras12345, 2019-07-25:
Hi, couldn't it be that your machine moves too fast (Z axis) while probing and loses some steps, so that when the tool retracts, it's somewhere else then the machine thinks it is?

@pacmancoder 11x12 = 132 points to probe, so 0.2 mm might be caused by low precision of your machine and/or losing steps while breaking when it touches the material.

yuranikspb, 2019-07-25:
> Hi, couldn't it be that your machine moves too fast (Z axis) while probing and loses some steps, so that when the tool retracts, it's somewhere else then the machine thinks it is?
>
> @pacmancoder [...]

No, because if I return to the coordinates of 0.0 and make a z probe, then I get a Z height of 0.004 (ie there no  loses steps). And if decrease the interpolation grid, the effect decreases

ondras12345, 2019-07-25:
Interpolation (see image on [wikipedia](https://en.wikipedia.org/wiki/Bicubic_interpolation)) could affect it if you have a false measurement in the heightmap, but it shouldn't be a problem if all the probed points are OK. You can inspect the [Candle interpolation function](https://github.com/Denvi/Candle/blob/master/src/utils/interpolation.h#L31) yourself, I don't have enough free time to do so right now. I think the function is called [here](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3579).

yuranikspb, 2019-07-25:
> Interpolation (see image on [wikipedia](https://en.wikipedia.org/wiki/Bicubic_interpolation)) could affect it if you have a false measurement in the heightmap, but it shouldn't be a problem if all [...]

i think need look to function where Interpolate value add to g code,  too.

ondras12345, 2019-07-25:
I would try this:
1. Open Candle settings, check _Show G-code program commands_ checkbox under _Console_
1. Create the heightmap and press _Send_
1. Wait until the machine gets to the position where it stops touching the material / goes below it, pause or stop the machine
1. Compare the command in the console with the one from the program file to see how much the interpolation function added to the original value - it should match the heightmap probed values

Another thing I'd try is probing after the job finishes to see if the offset isn't caused by the actual milling (losing steps, etc.).

yuranikspb, 2019-07-25:
I tried this:
1 Create the heightmap 140x200 (interpolation 40x40) pobe step 5mm wait for finish
2 go to x0y0  zprobe 0 (map probe 0)
3 go to 140,0 zprobe -0.352 (map probe in this point -0,341)
4 go to 140,200 zprobe -0.670(map probe in this point -0,658)

ondras12345, 2019-07-25:
> I tried this:
> 1 Create the heightmap 140x200 (interpolation 40x40) pobe step 5mm wait for finish
> 2 go to x0y0 zprobe 0 (map probe 0)
> 3 go to 140,0 zprobe -0.352 (map probe in this point -0,341)
> 4 go to 140,200 zprobe -0.670(map probe in this point -0,658)
> 5 create the heightmap 140x200 (interpolation 2x2) pobe step 5mm wait for finish
> 6 go to x0y0 zprobe 0 (map probe 0)
> 7 go to 140,0 zprobe -0.355 (map probe in this point -0,350)
> 8 go to 140,200 zprobe -0.673(map probe in this point -0,668)

Interpolation shouldn't matter if you don't apply the heightmap to a program.

yuranikspb, 2019-07-25:
I don't aplly, just compare height in table map and z probe manual

ondras12345, 2019-07-25:
> I don't aplly, just compare height in table map and z probe manual, but how size of interpolation affects the result, the test was conducted 3 times

Yes, but there is no way the interpolation could affect your results.

yuranikspb, 2019-07-25:
and there no step missing x0y0 always +- 0.0 height

ondras12345, 2019-07-25:
Size of interpolation grid shouldn'd have any effect during probing. You can change it after probing because it doesn't modify the probed points, it takes them as an input to calculate all the other points you haven't probed.

yuranikspb, 2019-07-25:
Then I have no idea what my problem is :(

ondras12345, 2019-07-25:
Is your material secured to the machine well? Can't it move and/or flex slightly while milling?

yuranikspb, 2019-07-25:
I check this many times, 
Moreover, this effect I noticed on my first small cnc with a plastic spindle holder , now I have another cnc with  portal of aluminum , and the effect is the same on different cnc.

ondras12345, 2019-07-25:
I'd try probing the machine bed directly to rule out the possibility of the PCB flexing.

> I even was tested for repeatability on the z-axis. Wrote g-code which is 1000 times raised the axle to 2mm and then lowered to 2 mm and the error was 0,008

The stepper isn't under load in these conditions, it might be worse if you try drilling into the material or probing it.

ondras12345, 2019-07-26:
Probing data is interpreted [here](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L1171) and here are the links to [`m_heightMapModel`](https://github.com/Denvi/Candle/blob/master/src/frmmain.h#L229)'s class' [header](https://github.com/Denvi/Candle/blob/master/src/tables/heightmaptablemodel.h) and [code](https://github.com/Denvi/Candle/blob/master/src/tables/heightmaptablemodel.cpp).
The [`updateHeightMapInterpolationDrawer`](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3193) function that is called while interpreting probing data doesn't modify `m_heightMapModel` at all.
Probing g-code program is generated [here](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3155) and it doesn't have anything to do with interpolation.

So the interpolation shouldn't really have any effect on the probing result.

ondras12345, 2019-07-26:
What _Heightmap probing feed_ do you use? (Main Candle settings)

yuranikspb, 2019-07-26:
z probe command( G21G91G38.2Z-30F35; G0Z1; G38.2Z-2F20)
map feed 20

ondras12345, 2019-07-26:
> z probe command( G21G91G38.2Z-30F35; G0Z1; G38.2Z-2F20)
> map feed 20

Probe command is just for the _Z-probe_ button, heightmap doesn't use it.
I use _Heightmap probing feed_ 10, I think you should try if decreasing it solves your problem.

yuranikspb, 2019-07-26:
I will change stepper for more powered, and decreas feed, and try again.

closed 2025-08-31 by github-actions[bot]