Denvi/Candle #247: Jogging does not work

hyudryu, 2018-09-09:
I'm on the most current version of Candle 1.1.7, and I'm unable to get the jog to work properly. I've been using grblcontrol 0.8.4 before without any issues but ever since getting Candle 1.1.7, I'm unable to move the stepper motors. I can turn on the spindle however, but the steppers won't move.

The software does control the steppers when I upload a .nc file, but I just can't manually move the motor

casnacaj, 2018-09-21:
Hi. I have the same problem. I build branch grbl_1_1 on ubuntu bionic. The jog does not work and the preview window is black and unusable.

Jodank, 2018-11-12:
Hi!
I use the same Candle and everything works great.
So, specify your CNC control board pls.
Do you updated the grbl firmware to the latest version? I assume it is the v1.1.

I didn't figured out any problem with the jogging in Candle 1.1.7 but I use it really frequently for my PCB projects.

darkglobe81, 2018-11-16:
Same here on win 7 (virtual machine on Mac)Grbl 0.8 jog worked properly but no tool and no visualization (white screen) but if I send the cn file the cnc works good on all the demo (iphone) file, but blind visualization.  then i’ve Installed Candle, now I can see the orange tool and coordinates, spindle rotates but jog doesn’t work.! Also i’ve uploaded a cn file, sent the file and motors worked so badly making ‘tuned sounds’ then stopped with error report. I’ve installed driver visual after an error (lack of .dll files). On windows 10 in a vaio laptop works good, on another tower with p4 and xp, same issues as win7.

Jodank, 2018-11-16:
Hmmm...
Interesting, I didn't checked on Linux, but I already can confirm does my CNC 2418 is working great on Win XP x32, Win7 x64, Win10 x64 with the latest version of Candle and the Grbl firmware v1.1.
I'm really sure, I would never use the tool on a VM.
I have bad experiences with VM solutions and real hardware tunneling through them.

darkglobe81, 2018-11-16:
Well my other option is to run in on a 2005 pentium 4 with xp. But as  I said the jog works but nor the orange gear on coordinates an the file image doesn’t appear. If I push send the machine do the job. Would you please remember or suggest the right setup in order to working properly on Xp old P4 hp compaq?
Thanks for your feedback!
N

Inviato da iPhone

Jodank, 2018-11-16:
Actually, you may have some issue with the graphical visualization as I
understand.
That lead me to some problem with the OpenGL section.
I think Candle use the OpenGL API to render the vector graphics.
Some graphics card have trouble with OpenGL, you should try to install
the original driver for your OS for your graphics card.
Also check what version of OpenGL required for Candle and check what is
your installed version too.

darkglobe81, 2018-11-16:
I use native graphic, no separate boards. I’ll try to download OpenGL from windows anyway and install it in case i lost something, ( I performed a bios hard reset and coin cell swap cause the old desktop didn’t booted).
Actually even the win7 VM was lacking the OpenGL exe and once I put it on win graphics appeared but with the jog problem....
Too may head ache 😖 if only there was the camcad for OS X ... ☹️
Thank you 🙏🏼 

Inviato da iPhone

Jodank, 2018-11-16:
Let us know about your progress...
I hope you will solve your problem quickly.

darkglobe81, 2018-11-18:
Ok i don’t know exactly what I’ve done, I quitted with VM on my Mac and I restored an old P4 HP compaq 2,6ghz 2,5 gb and win xp.
I took all the folders with Grbl 0.8 l, candle, candle2, grb x32, 64, drivers ALL THE STUFF CAM RELATED in a single desktop folder ...and surprisingly now it works perfectly!
This makes some sense cause each version of Grbl was lack something, jog or graphic or connecting...so jocking I created a Frankenstein folder thinking what the hell, all the things the program need is here somewhere in the folder! :D I’m shocked ahahah

Inviato da iPhone

MeJasonT, 2018-11-18:
Just out of curiosity
Is this related to your issue  Visualisation issue #245
[https://github.com/Denvi/Candle/issues/245](url)

darkglobe81, 2018-11-18:
247 but I surfed a lot on the subject so probably it’s related

Inviato da iPhone

lexelby, 2019-04-20:
In my case, I had to enter a value into the "feed" box in the Jog panel before it would actually do anything.  If it's blank, jog buttons are ignored.

shahalam32, 2022-07-14:
I have same problem, and also spindle doesn’t response.

closed 2025-08-31 by github-actions[bot]