Denvi/Candle #251: Use Heightmap checkbox disabled, why?

Jodank, 2018-10-21:
Ok, I was trying the Heightmap option in this version of Candle, whenever I finish the measuring of the 
pcb everything is fine, but the checkbox "Use Heightmap" is non-stop gray.
Why?

Will this function automatically correct the Z depth when I try to mill my pcb after I measured the surface 
deviation or I have to do some other thinks too?

What I doing wrong?
Thanks.

Jodank, 2018-11-11:
Problem solved.

brawd, 2019-08-12:
Being as I have the same problem and I am a rank beginner at all this can you tell us how it was solved please.

fodormarton, 2021-05-03:
> Being as I have the same problem and I am a rank beginner at all this can you tell us how it was solved please.

You need to load G-code before you can use it. The checkbox still remained disabled for me, but I could run the hightmap function by pressing the 'Create' button.

closed 2025-08-30 by github-actions[bot]