Denvi/Candle #252: Work and machine coordinates does not actualize when G92 is run

ondras12345, 2018-10-27:
Hi,
I just actualized my candle to 1.1.7 and the coordinates text boxes stopped to work. When moving the machine, it normally shows the position. But when I use the `G92` command to set the coordinates to zero, it does not actualize. When I move the machine then, it shows actualizes and shows me the bad position (only adds the movement distance to the bad values).
Eg. when i run:
1. `G0 X10`
2. `G92 X0`
3. `G0 X5`

I get:
1. X=10
2. X=10 while the work coordinates should be X=0
3. X=15 while the work coordinates should be X=5

[image]

My settings are:
`$0=10 (step pulse, usec)`
`$1=25 (step idle delay, msec)`
`$2=0 (step port invert mask:00000000)`
`$3=5 (dir port invert mask:00000101)`
`$4=0 (step enable invert, bool)`
`$5=0 (limit pins invert, bool)`
`$6=0 (probe pin invert, bool)`
**`$10=3 (status report mask:00000011)`**
`$11=0.010 (junction deviation, mm)`
`$12=0.002 (arc tolerance, mm)`
`$13=0 (report inches, bool)`
`$20=0 (soft limits, bool)`
`$21=0 (hard limits, bool)`
`$22=1 (homing cycle, bool)`
`$23=3 (homing dir invert mask:00000011)`
`$24=25.000 (homing feed, mm/min)`
`$25=500.000 (homing seek, mm/min)`
`$26=250 (homing debounce, msec)`
`$27=1.000 (homing pull-off, mm)`
`$100=800.000 (x, step/mm)`
`$101=800.000 (y, step/mm)`
`$102=800.000 (z, step/mm)`
`$110=1000.000 (x max rate, mm/min)`
`$111=800.000 (y max rate, mm/min)`
`$112=500.000 (z max rate, mm/min)`
`$120=10.000 (x accel, mm/sec^2)`
`$121=10.000 (y accel, mm/sec^2)`
`$122=10.000 (z accel, mm/sec^2)`
`$130=160.000 (x max travel, mm)`
`$131=100.000 (y max travel, mm)`
`$132=45.000 (z max travel, mm)`

This happens on both of my computers (Windows 10 and Ubuntu 18.04). I was using Candle 0.9 before and it worked much better.
My machine is a cheap Chinese 1610 cnc with Grbl 0.9j.

ondras12345, 2018-10-27:
I used putty to see the `?` response, it looks like this: `<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
`. Isn’t that the problem? According to [https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface](url), the grbl 1.1 output should look like this: `<Idle|MPos:0.000,0.000,0.000|FS:0.0,0>`. Is the new Candle version still compatible with Grbl v0.9j?

Denvi, 2018-10-27:
Hello, Candle 1.0.x was the latest versions which works with GRBL v0.9.

closed 2018-10-27 by ondras12345