Denvi/Candle #253: STATUS PROBLEM with 3018, woodpecker, Grbl controller,port opened???

winson2769, 2018-10-29:
Status is reading "Port Opened" & will not allow Gcode to run or jogging.to be done.
Need status to read IDLE etc.
Command line in GRBL control shows lines of Gcode as sent, although when file is exited it reads "File sending in progress.
Problem seem to start after system was powered off to stop ran away motor.
USB port works with other items such as mouse, or flash drive.
Suggestions or help appreciated.

Jodank, 2018-11-11:
Hi!
The "Port opened" should say your com port is open.
You have to check why is your selected com port open before you open it through Candle.
If your com port is opened before you start Candle and it really left open, then no command can be sent to the com port through Candle, that is a normal situation.
so, you cant send a jog or whatever command to the cnc.

An abnormal situation in your case is, why is your com port left open?

winson2769, 2018-11-12:
hi thanks for feedback, for the first time using grblcontrol 0.8 and all running well. but after shutdown pc and cnc machine, the next time cant connect to pc and  always show open port.
is that anyway to close the port or download anykind of software to close it?
thanks.

Jodank, 2018-11-12:
Hmmm...
I really can't tell you what is the problem in your situation, I also used the grblcontrol app and worked for me fine. I got some problems but I switched quickly to the newest version which is the Candle app.
To use Candle you need to upgrade the firmware in your hardware to grbl1.1.
That's really easy and you have a fine explanation how-to.

So, in your case, you should check if your grblecontrol app opens the correct com port.
Do you have any other devices on a com port attached on the same PC?
If yes, maybe grblcontroll tries to open a com port what is already used by other hardware.

When you start grblcontrol, can you use the jog function?
I mean, can you move the machine XYZ axes?

Jodank, 2018-11-13:
Ok, here is the wiki page with explanation how to upgrade to v1.1:
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

**WARNING!**
Before you upload through the XLoader the .hex file to your board, please make a backup of your
current eeprom content, after the update you have to check the new configuration of your machine and 
also, have to setup the parameters of the machine in your eeprom.

To read out the eeprom use the command $$ in the command line near the Cosol windows.
You should get something like this:

> $1=25
> $2=0
> $3=5
> $4=0
> $5=0
> $6=0
> $10=3
> $11=0.010
> $12=0.002
> $13=0
> $20=0
> $21=1
> $22=0
> $23=0...

Select and copy/paste it to notepad and save it to a vey safe place.
This setup is a must have for your machine.
Without setting up through this, your machine will have incorrect operations like measurement 
precisions, axes are inverted when you move the axes in some desired dirrection, the axes would move oposite etc...

I can fully use the Cnadle and the older version of grbl control software with my cnc, 
almost all functions are working well.
I can't only control the spindle speed because I don't have a pwm controller on my board.
I can only power on and off my spindle.

winson2769, 2018-11-15:
hi can you pls give me the grbl hex file to flash?

Jodank, 2018-11-15:
Hi!
Here is a good link what you should first read before you start to upgrade:
https://sienci.com/2018/07/04/firmware-update-grbl-1-1g/

[Grbl 1.1 firmware.zip](https://github.com/Denvi/Candle/files/2585729/Grbl.1.1.firmware.zip)

I hope I helped you.
My best regards.
Chris.

winson2769, 2018-11-18:
thank you so much for your kind help, its seem like doing well. thanks.

winson2769, 2019-01-02:
hi sir  after the update the firmware ,the new configuration of my machine is totally run out spect and can u teach me how to setup the parameters of the machine in your eeprom. my machine is 3018 cm.thanks.

winson2769, 2019-01-02:
this is my parameter now
$$ < $0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
ok

ondras12345, 2019-01-02:
The parameters can be set by typing `$x=value` (see https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration)

jackzter, 2019-02-08:
you must unplug offline controller to get it to work with computer

jackzter, 2019-02-08:
for the 3018 grbl 1.1 woodpecker you can't have the offline controller connected to board while sending code with computer

elm225cr, 2020-06-19:
Hello.. 

I´m new to this grbl subject and I´m struggling with my 3018 cnc router. 

It comes with a Woodpecker 3.1 grbl board. 

I have followed all the steps correctly:

1. Install driver CH340
2. Identifiy COM port and baud rate
3. Open grbl software (Candle 1.1.7)
4. Update settings (COM and baud rate)

But then I get a message in the STATE panel that says reads **" Status: Port opened"**

I can´t move forward from there (see file attached)
[image]

PLEASE HELP!!!

lucasminadeo, 2020-07-24:
Hola elm225cr, me estaría pasando lo mismo que a vos, pudiste resolverlo?

> Hola..
> 
> Soy nuevo en este tema de grbl y estoy luchando con mi enrutador cnc 3018.
> 
> Viene con un tablero de carpintero 3.1 grbl.
> 
> He seguido todos los pasos correctamente:
> 
> 1. Instale el controlador CH340
> 2. Identificar puerto COM y velocidad de transmisión
> 3. Abra el software grbl (Candle 1.1.7)
> 4. Configuración de actualización (COM y velocidad de transmisión)
> 
> Pero luego recibo un mensaje en el panel ESTADO que dice: **"Estado: puerto abierto"**
> 
> No puedo avanzar desde allí (ver archivo adjunto)
> [image]
> 
> ¡¡¡POR FAVOR AYUDA!!!

Hola elm225cr pudiste resolver tu problema? Porque a mí me estaría pasando lo mismo! Gracias y saludos

mariadelaosaa, 2021-12-19:
Hola! me pasa exactamente lo mismo que a elm225cr. Podéis ayudarme?

astetner, 2022-09-11:
Did @elm225cr find a solution? About a year later and I have the same issue...

astetner, 2022-09-11:
¿@elm225cr encontró una solución? Un año después y tengo el mismo problema...

NyarlathotepBS, 2023-02-22:
buen día, ando intentando utilizar mi woodpecker v3.4 pero al conectarlo no prende el led indicador azul de comunicación y en el  candle me da port open, intente cambiar el baud rate pero sigue sin dejarme trabajar con la tarjeta alguna idea?

closed 2025-08-30 by github-actions[bot]