Denvi/Candle #255: Problem with resetting COM port!

Jodank, 2018-11-11:
Hi!
I have a strange situation.
When I open a file for working with, and I start to calibrate my cnc coordinates,
unpredictable the machine just resets the COM port.
I'm hearing a small "tick" on the motors and the actual XYZ coordinates are
reset to 0 and so I lose the coordinates I set.

This issue is not happening until the CNC is milling, engraving, or whatever...

What can happen here?
Any idea?
Thanks.

Jodank, 2019-01-02:
> Check your wiring. There could be some inductions from the motor signals
> and/or the spindle power. I had to disable hard limits on my machine
> because they were stopping the milling process. It could be some induction
> from the wiring to the controller's board. What is your controller board?
> You could try adding a capacitor between 5v and GND (eg. 10uF - 47uF).
> This situation could also be caused by vibrations. Try checking the
> connectors and solder joints - there could be a cold solder joint.

I think this kind of end switching think is really bad.
I was trying with caps but didn't solved the problem as expected.
It was unpredictable when it would be triggered.

For now I didn't use this function on my cnc, maybe I will build a
debouncing circuit and then
I will give it a try.

My suggestion is to make a hardware switch debouncing circuit and then use
it with the controller.
I think this would be a nice IC to solve the problem really:
https://datasheets.maximintegrated.com/en/ds/1896.pdf

My best regards.

ondras12345, 2019-01-02:
I tried the optocoupler method because it was suggested in the wiring limit switches tutorial.
The 555 timer forms a schmidt inverter with thresholds in 1/3 and 2/3 of the supply voltage. It works without problems.

ondras12345, 2019-01-02:
> When I open a file for working with, and I start to calibrate my cnc coordinates,
> unpredictable the machine just resets the COM port.
> I'm hearing a small "tick" on the motors and the actual XYZ coordinates are
> reset to 0 and so I lose the coordinates I set.

The hard limits should not cause this behaviour. I guess the ATmega is resetting because of BOD (brown-out detection), or the reset pin being pulled low (you can capture that using an digital oscilloscope).

ondras12345, 2019-01-02:
For example the woodpecker controller board I have has both reset and 5v pins on the main header, so you can probe it easily. More info about the pinout here https://github.com/gnea/grbl/issues/123.
[image]