Denvi/Candle #257: Suggestions for next versions: Keyboard

the-powl, 2018-12-01:
Great program and very clean UI! After using it for a while, I can say I really like to work with it! But I still miss some features. So here are some suggestions to implement in future releases:

1) Edge/Centerfinding. Add the ability to do probing sideways so you can find the edges of a workpiece. Make a button for easily setting X and Y-coordinates separately while considering the tool-diameter. This way you can easily use one of these readily available aluminium edge finding tools. Maybe even add the ability to probe 3 points within or outside a circle and calculate the corresponding centerpoint. Create an option for automatic probing or manual probing (if you are using a manual rotating centerpoint finding tool and work with nonconductive materials)

2) Automatic workpiece orientation. As an adition to the edgefinding ability, make a 3-way edgefinding so one can not only find the edges of a rectangular workpiece but also the angular orientation. This would be very useful for double sided milling where it's very important to find the exact position of a workpiece again after turning it around.

3) Manually assignable keyboard shortcuts. Create a menu where you can assign specific program actions to self-assigned keyboard shortcuts so you can use whatever input device you want. Small bluetooth keyboards for example oder custom made arduino button pads.

4) More Usercodes. Add the ability to define a second set of usercodes or maybe even an arbitrary number of them.

These are just some suggestion could be - in my personal humble opinion - helpful in the daily work with Candle. So you can choose which ones you like and find useful and therefore want to implement. Keep up your good work!

ondras12345, 2018-12-27:
It would also be nice to have ability to **change zero XY and Z buttons' commands**.
Since I use homing, I prefer to zero my work coordinates using **coordinate systems**. (`G10 L20 P1 X0 Y0 Z0` - changes actual coordinates in `G54 `to 0 0 0 More info: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2). I think it is better because these changes are stored in the EEPROM of my controller, so I can continue the cut after power loss, etc.
I am using the user command buttons to do it now, but it would be better to just modify the dedicated UI buttons.
It would be enough if you gave us the ability to change this in the config file, you don't have to create a form for it.

Another suggestion is to map the user commands to the keyboard, eg. F1-F4.

ondras12345, 2018-12-27:
> 3. Manually assignable keyboard shortcuts. Create a menu where you can assign specific program actions to self-assigned keyboard shortcuts so you can use whatever input device you want. Small bluetooth keyboards for example oder custom made arduino button pads.

I use **luamacros** for this. It can map any key on the keyboard to do anything (run sw, type something), and also distinguish between keyboards - you can change the behavior of just the bluetooth keyboard while keeping your main one unaffected.
It can only remap actions that have a key assigned though. It can't be used to assign any action you want to a key.

http://www.hidmacros.eu/forum/viewtopic.php?t=241
https://github.com/me2d13/luamacros

dremin1984, 2019-03-10:
I'd love to have a candle work in raspberry PI3!!!!!

closed 2025-08-30 by github-actions[bot]