Denvi/Candle #259: Trying set up Z probe on cnc?

unioncreek, 2018-12-30:
I have a cnc engraver and using Candle 1.1.7.  I'm trying to set up my Z probe.  I'm using a piece of steel sheet metal that measures 0.025".  When I use the "Z Probe" button is will stop when it contacts the sheet metal.  I'm trying to find when I can enter the 0.025" measurement.  Looking for any help I can get, totally a newbie when it comes to this.

Thanks,
Bob

ondras12345, 2018-12-31:
> I'm trying to find when I can enter the 0.025" measurement.

Hi, are you trying to compensate for the material's height? (Make the material's top Z0?) Then, you can press the button zero Z. It uses `G92` command (http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g92) to set the offset.
You can also do it with coordinate systems (http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2 and http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l20), but you'd have to type the commands to the console that way. There are no buttons to work with coordinate systems. 
What is your GRBL version? I had some problems while using Candle 1.1.7 with my GRBL v0.9j machine (#252). In short, Candle versions above v1.0.x are not compatible with GRBL v0.9.

ondras12345, 2019-01-01:
The GRBL version gets printed into the console when the machine starts.
Sorry, but I don't understand what are you trying to do.
Do you want to measure the plate's height using the CNC?
Do you want the plate's top to be Z0? (Z-2 then means 2" into the plate)
[image]

Or, do you have a non-conductive material and you are trying to probe it? (so you need the plate's top to be Z0.025 --> material's top Z0) Then, you can use `G92 Z0.025` (see G92 command reference in my previous post) to set the tool's position to Z0.025.
[image]

How did you change the units to inches? Did you use the machine's setting `$13=0 (report inches, bool)`, or did you use the G20 command? (http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g20-g21). 
I use millimeters, so I don't have any experience with imperial units.

unioncreek, 2019-01-01:
ondras12345,

Example #2 is what I'm trying to do. I'm cutting nonconductive material.

Thanks for helping me out.  I figured the coordinates display out.

Bob

closed 2019-01-01 by unioncreek