Denvi/Candle #260: candle keeps disconnecting from cnc during job

leedash, 2019-01-04:
Hello.  I am very new at this and have searched for and failed to find a solution

After a very short time into any test job my candle displays "not connected" and my cnc just freezes with the spindle still going.  If i disconnect and reconnect the usb and reset it will happen again

The max amount of steps I've gotten to is about 140.  What could be causing this please?

I've left the program open for over an hour and it sits at "idle" happily

It is a 1610 chinese machine

ondras12345, 2019-01-04:
Try this: https://github.com/Denvi/Candle/issues/255

ondras12345, 2019-01-04:
BTW, for the 1610 machine I suggest you not to tighten the screw holding the spindle too much. Otherwise, the machine can end up like mine:
[image]

ondras12345, 2019-01-04:
Do you have the heatsinks attached to the controller board? They should be on the three stepper control plug-in boards. The chips are getting hot even with the heatsinks... but it shouldn't cause the machine to disconnect. The steppers could stop, but the microcontroller should run normally.

leedash, 2019-01-04:
Hi. Thanks for the reply.  That link is a bit too much for me :-/  I'm not that techie

Yeah I am not that confident in the plastic.  I had to heat it to get it into shape before i could fit the spindle in so it is already very tight!

Heatsinks are on yeah.  Could it be shorting somewhere?  I've use the nylon washers.  Interference from othe rcables?

leedash, 2019-01-04:
Update.  I tried two other test programs.  These fail on the same instruction.  That's not a co-incidence?

ondras12345, 2019-01-04:
What instruction does it fail on? Can you take a photo of your wiring?

ondras12345, 2019-01-04:
You can also try running the code in check mode (machine doesn't move, but responds normally) Maybe it's a GRBL software problem.

leedash, 2019-01-04:
Z instructions.  I ran it a few more times to be sure and it doesn't stop on the exact one but it is always a Z

[image]

leedash, 2019-01-04:
Running in test mode now and it's gotten way further and still going

ondras12345, 2019-01-04:
Take a look at the Z controller board. Isn't the heatsink shorting the pins? Also, take a look at the bottom of the controller board and check the main power connector's solder joints. Aren't they cracked? It could be some sort of issue caused by vibrations.

What power supply are you using? Can it provide enough current?

leedash, 2019-01-04:
I appreciate this help.  I moved the heat sink a little.  Can't see any cracks

If it is a vibration issue how to I solve it?  Do I need a new board?

leedash, 2019-01-04:
The power supply came with it.  LiteOn EPS-5 240v

ondras12345, 2019-01-04:
Does it print anything to the console when it fails? Do you have to disconnect the PC and reconnect it, or does pressing the reset button on the board help?

ondras12345, 2019-01-04:
Try running the same code while having the spindle above the material. Does it fail too?

leedash, 2019-01-06:
Hi.  Thanks very much for the help.  Turned out it seems to be an overheating problem.  i moved it to another room and opened a window and now it runs.  Had no idea it'd be that sensitive

It wasn't printing anything to the console no.  I had to disconnect and re-connect the usb from the computer to make the spindle stop running

Next job is getting it all aligned properly etc!

closed 2019-01-06 by leedash

ondras12345, 2019-01-06:
I asked about the USB disconnection because if the reset button doesn't help (the spindle will stop but the machine won't reconnect), it means that you have some problems with the CH340 IC that is used to handle the USB. 
[image]

Overheating could cause the steppers to stop, but it shouldn't cause the controller to disconnect.