Denvi/Candle #262: Polling of Grbl

MeJasonT, 2019-01-30:
Hi Denvi,
is there any way to change the rate at which the GUI polls GRBL for information.
I'm looking for around 200ms but Candle is super rapid.
There doses not appear to be an option in settings

I'm having to go back to GRBL Panel as it is set to 200ms by default
Reason for asking:-
I have written and designed a system that listens to the echoed responses from grbl by sniffing the TX string and transmitting it over fibre to control a Variable Frequency Drive for my AC 3 phase spindle motor. (Its running an old industrial Milling machine).

I have noticed the site isn't updated too often, are you still supporting candle - please say yes.

Denvi, 2019-01-30:
Hello.
Check for "Status query period" in "Settings -> Machine information" section. It's set to 40 ms by default.

MeJasonT, 2019-01-30:
Cheers buddy,
It was obvious when you pointed it out to me.
many thanks
Jason
ps. do you have any plans to upgrade or add more functions, the look and feel of the GUI is superb and i would love to see it go further.

closed 2019-02-14 by MeJasonT