Denvi/Candle #263: Is it possible to change to the negative coordinate system

MeJasonT, 2019-02-14:
Is it possible to change to the negative coordinate system, i have been trying to figure out how i can get my work 0,0 position to be far right corner instead of bottom left of machine axis.  Similar to professional milling machines.

ondras12345, 2019-02-14:
Hi, you can use homing to do that (if you have your homing switches installed) https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle#homing-direction

Or you can manually jog the machine and then press the reset button on your controller board (not the one in the Candle interface). The place where you reset your controller now becomes 0,0.

MeJasonT, 2019-02-14:
Acquiring the 0,0 is not the issue i'm having.  I opened a file and selected it to run, the machine immediately ran off to the bottom left corner of the bed to start the job.  My machine is set to home in the top left corner as is normal practice.  I don't even know why i haven't noticed this before.

ondras12345, 2019-02-14:
Can you post the first lines of the code?
You can set your work coordinates 0,0 using the G92 command http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g92

MeJasonT, 2019-02-14:
Code as requested

( Grbl POST for Elliot VMC )
( face25mm bar 2/14/2019 12:08:07 PM )
G21 G90
G0 Z20.0
( Profile1 )
T305 M6 (Part1\Profile1\10.00mm 3 flute End Mill\10.0)
G17
M3 S1016
G0 X5.0 Y4.0
G0 Z1.0
G1 F150.0 Z-0.4
G1 F85.0 X105.0
G0 Z20.0
G0 X5.0 Y8.0
G0 Z1.0
G1 F150.0 Z-0.4
G1 F85.0 X105.0
G0 Z20.0
G0 X5.0 Y12.0
G0 Z1.0
G1 F150.0 Z-0.4
G1 F85.0 X105.0
G0 Z20.0
G0 X5.0 Y16.0
G0 Z1.0
G1 F150.0 Z-0.4
G1 F85.0 X105.0
G0 Z20.0
G0 X5.0 Y20.0
G0 Z1.0
G1 F150.0 Z-0.4
G1 F85.0 X105.0
( TrochoPock1 )
S1000
G0 Z20.0
M5

Cheers
J.

ondras12345, 2019-02-14:
Sorry, I thought you want to set the machine coordinates, not the work coordinates. 
You want the `G0 X5.0 Y4.0` line to move to X5 Y4 of the material, not the machine, right?
Then you can use the G92 command or coordinate systems http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l2 to set the work 0,0 to match the material's corner.

MeJasonT, 2019-02-14:
I'm using GRBL, i will have to see if i can use the same commands as the Linuxcnc.
I will have a try when i'm back at my machine tomorrow.
Looks like i'm learning something new.

Thanks for your help its much appreciated

ondras12345, 2019-02-14:
I used the Linuxcnc just as a reference. I tested these commands and they are compatible with GRBL 0.9

MeJasonT, 2019-02-14:
Brilliant, thanks.
will try tomorrow

closed 2019-03-08 by MeJasonT