Denvi/Candle #264: Latest GRBL 0.9 revision

cedricp, 2019-02-15:


Denvi, 2019-02-17:
Hello, grbl_1_0 branch is right before merge: https://github.com/Denvi/Candle/tree/grbl_1_0.
App can't support 0.9 and 1.1 firmwares simultaneously right now.

MeJasonT, 2019-02-17:
Only change that springs to mind is $10 setting in GRBL.  It has been covered in many posts as an issue for Candle/GRBL not displaying cutter moves on the GUI

cedricp, 2019-02-17:
Ok, thank you, what are the main differences between 1.1 and 0.9 ? I searched the web, but didn't find useful information. I'm an experimented C/C++ developer, and I'd be glad to help.

MeJasonT, 2019-02-17:
Initially there were hardware changes
[https://github.com/grbl/grbl/wiki/Connecting-Grbl](url)
and the status report feedback messaging changed, the status report used to be configurable so that only required machine conditions were transmitted. ie. setting 19 to $10 used to show the limit switches in the reply with loads of other stuff.  In 1.1 $10 can only be 1,2 or 3.  if you have a look at the fixes associated with the the releases for GRBL then you will see what has changed, this should give you an idea of what would need to be altered in order to run with GRBL 0.9.  Why do you want to - is it not possible for you to upgrade your GRBL firmware to 1.1.

closed 2025-08-30 by github-actions[bot]