Denvi/Candle #269: Сборка проекта

shatohingv, 2019-03-12:
Добрый день. пытаюсь пересобрать проект на C# но как то не выходит. Половину файлов не видит (я кстати и руками эти файлы не нахожу) а половину видит. Может я что то не так делаю?

yuranikspb, 2019-03-29:
Build requirements:
Qt 5.4.2 with MinGW/GCC compiler

Написанно же
я в qt делал себе изменения добавил калькулятор шага автоматический и собрал свою версию

DEST1981, 2023-03-19:
Народ, подскажите как собрать проект. Бьюсь уже не одну неделю не получается. Никогда с QT не работал. Перепробовал кучу комбинаций, везде различные ошибки вылезают.

mar0x, 2023-03-19:
@DEST1981 , уточните из какой ветки собираете и под какой OS ?
Описание конкретных проблем и цитирование ошибок поможет сформулировать для Вас конкретные рекомендации.

DEST1981, 2023-03-19:
Ветка экспериментальная. В ней Денис вносил изменения которые мне были нужны. Операционка Windows10.
Установлено: qt-5.4.2-x86-mingw510r0-dw2-rev0 + qt-creator-3.5.0-x64
Попытка сборки выдаёт: Project ERROR: Unknown module(s) in QT: uiplugin
Вероятно не хватает библиотек.
Вероятно не та версия QT. Большой вопрос где взять нужную версию.

mar0x, 2023-03-19:
Мне удавалось собрать экспериментальную ветку с некоторым количеством исправлений и огромным количеством warning'ов под Qt 5.15.2 + MSVS 2019. Исправления сложил вот сюда: https://github.com/mar0x/Candle/tree/Experimental-fixes

XRay3D, 2023-03-19:
https://github.com/XRay3D/Candle я написал сборщик CMake для актуальной сборки.

DEST1981, 2023-03-20:
А можете инструкцию набросать как им пользоваться. 
Удалось собрать qt5.5. Но собралось кривовато, куча варнингов и часть функционала не работает.
Собственно я чего упираюсь. В сети есть проект fluidnc, это контроллер чпу на esp32. Проект развивается, совместимость с grbl есть но не полная. Candle ИМХО лучший софт для хобийного станка, но с fluid работает не корректно. Его бы допилить немного.......

mar0x, 2023-03-20:
Боюсь, что у экспериментальной ветки нет будущего из-за зависимости от [Qt Script](https://doc.qt.io/qt-5/qtscript-index.html) и [qtscriptgenerator](https://github.com/phbaer/qtscriptgenerator) . Основная ветка намного стабильнее и совместимость её с новыми версиями вызывает меньше опасений.

kossnikita, 2023-08-20:
Собираю https://github.com/mar0x/Candle/tree/Experimental-fixes под Qt 5.15.2 + MSVC2019 32bit и ловлю ошибку
```
	link /NOLOGO /DYNAMICBASE /NXCOMPAT /OPT:REF /INCREMENTAL:NO /DLL /SUBSYSTEM:WINDOWS /OUT:..\..\..\bin\customwidgets.dll @C:\Users\KOSSNI~1\AppData\Local\Temp\customwidgets.dll.8852.61640.jom
   Создается библиотека ..\..\..\bin\customwidgets.lib и объект ..\..\..\bin\customwidgets.exp
jom: D:\build-candle-Desktop_Qt_5_15_2_MSVC2019_32bit-Release\src\scriptbindings\qtbindings\qtscript_custom\Makefile.Release [..\..\..\..\bin\script\qtscript_custom.dll] Error 1181
jom: D:\build-candle-Desktop_Qt_5_15_2_MSVC2019_32bit-Release\src\scriptbindings\qtbindings\qtscript_custom\Makefile [release] Error 2

...

	link /NOLOGO /DYNAMICBASE /NXCOMPAT /OPT:REF /INCREMENTAL:NO /DLL /SUBSYSTEM:WINDOWS /OUT:..\..\..\..\bin\script\qtscript_custom.dll @C:\Users\KOSSNI~1\AppData\Local\Temp\qtscript_custom.dll.26936.25860.jom
LINK : fatal error LNK1181: не удается открыть входной файл "customwidgets.lib"
```

При повторном запуске сборки, заканчивается успешно. Но приложение крашится на `qtscript_QDockWidget.cpp:215`
Дебагер выдает

```
:-1: error: Debugger encountered an exception: Exception at 0x5aaf967c, code: 0xc0000005: read access violation at: 0x6f446f4e, flags=0x0 (first chance)
```

kossnikita, 2023-08-20:
С https://github.com/Denvi/Candle/tree/Experimental дела еще хуже. Он не собирается никак ни под MSVC, ни под MinGW.

В Qt 5.15.2 MinGW все время норовит подставить флаги от MSVC

```
mingw32-make[4]: Entering directory 'D:/build-candle-Desktop_Qt_5_15_2_MinGW_32_bit-Debug/src/scriptbindings/qtbindings/qtscript_core'
g++ -c -fno-keep-inline-dllexport /MP /GS /Zi -O2 -Wall -Wextra -Wextra -fexceptions -mthreads -DUNICODE -D_UNICODE -DWIN32 -DMINGW_HAS_SECURE_API=1 -DQT_QML_DEBUG -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SCRIPT_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -ID:\Candle\src\scriptbindings\qtbindings\qtscript_core -I. -ID:\Candle\src\scriptbindings\qtbindings\qtscript_core -ID:\Candle\src\scriptbindings\qtbindings\qtscript_core\include -IC:\Qt\5.15.2\mingw81_32\include -IC:\Qt\5.15.2\mingw81_32\include\QtWidgets -IC:\Qt\5.15.2\mingw81_32\include\QtGui -IC:\Qt\5.15.2\mingw81_32\include\QtANGLE -IC:\Qt\5.15.2\mingw81_32\include\QtScript -IC:\Qt\5.15.2\mingw81_32\include\QtConcurrent -IC:\Qt\5.15.2\mingw81_32\include\QtCore -Irelease -IC:\Qt\5.15.2\mingw81_32\mkspecs\win32-g++  -o release\plugin.o D:\Candle\src\scriptbindings\qtbindings\qtscript_core\plugin.cpp
g++: error: /MP: No such file or directory
g++: error: /GS: No such file or directory
g++: error: /Zi: No such file or directory
mingw32-make[4]: *** [Makefile.Release:886: release/plugin.o] Error 1
```

В MSVC к ветке [mar0x/Experimental-fixes](https://github.com/mar0x/Candle/tree/Experimental-fixes) добавляется ещё ошибки

```
qtscript_QSocketNotifier.cpp(56): error C2766: явная специализация; уже имеется определение "QMetaTypeId<QSocketNotifier::Type>"
```

mar0x, 2023-08-20:
@kossnikita Приведите пожалуйста полностью все команды, которые Вы запускаете и полный вывод.

Примеры команд для успешной сборки можно найти в actions (напр. https://github.com/mar0x/Candle/actions/runs/5708999838)

kossnikita, 2023-08-21:
@mar0x Эти команды видимо для мастер ветки, поскольку в src нет CMake в экспериментальной ветке.

mar0x, 2023-08-21:
@kossnikita Если Вам нужна помощь со сборкой, приведите пожалуйста команды, которые Вы выполняете (с указанием ветки). Для воспроизведения Ваших ошибок хорошо бы знать также версию компилятора.

В разных ветках свои action'ы. Ссылка, которую я приводил ранее (https://github.com/mar0x/Candle/actions/runs/5708999838/job/15467228318) - это сборка ветки fixes. Ветка Experimental-fixes пока собирается qmake. CMake сборка в процессе доработки.

kossnikita, 2023-08-21:
Я Experimental-fixes собираю qmake'ом Qt 5.15.2 mingw810_64.

```powershell
${env:path}='C:\Qt\5.15.2\mingw81_64\bin;C:\Qt\Tools\mingw810_64\bin'
qmake -r  .\candle.pro
mingw32-make all -j20
```

Сборка завершается успешно, с тонной ворнингов. Но приложение сразу крашится без вывода ошибки.

mar0x, 2023-08-21:
@kossnikita ошибки линковки и их отсутствие при повторном запуске могут объясняться использованием ключа `-j20` и отсутствием зависимостей в makefile'ах, сгенерированных qmake. Попробуйте не использовать `-j` совсем (даже если у Вас простаивают 19 ядер).

closed 2025-08-30 by github-actions[bot]