Denvi/Candle #271: Карта высот. Несанкционированный подъем шпинделя.

Kurdl, 2019-03-20:
Имею китайский простейший 3-х координатный станок под управлением GRBL (v. 1.1).
Я пытаюсь снять карту высот в программе "Candle" (v. 1.1.7 Win)
Все настройки программы - по умолчанию.
НО! Нажатие на кнопку "Зонд" вызывает неожиданную реакцию - шпиндель несется вверх до упора в концевой датчик.
После этого станок блокируется и, понятное дело, миссия не выполняется.
При этом все остальные функции этого станка с помощью "Candle" исполняются корректно.
Даже нажатие кнопки "Z-щуп" вызывает нормальное поведение - щуп с заданной подачей опускается до контакта с поверхностью, останавливается и повторяет цикл на медленной подаче.
Я сравнил свои настройки "Candle" и GRBL с настройками "Candle" и GRBL у владельца такого же станка, который давно и успешно использует функцию "Карта высот" этой программы.
Единственное отличие в настройках: $21=1 (Жесткие границы, логический), т.к. в моем станке установлены концевые датчики.

MeJasonT, 2019-03-20:
It sounds like the probe direction is inverted
here are a few things i came across which may help
[https://github.com/grbl/grbl/issues/1126](url)
[https://github.com/vlachoudis/bCNC/issues/243](url)
There are 3 settings which handle direction of axis $3, $23 & $6 will play a part in the probe direction to a certain point.
If your settings and wiring are exactly the same on both machines, copy the config.h from the working machine and copy+upload it to the non working machine.  This might correct the problem.  Its not 100% guarantee that both machines are wired identically so you will need to make absolutely certain they are.
Just because you have bought a stepper motor that appears to be the same as one from another seller doesn't mean the four wires/coils are in the same order.  I don't think this is likely as you will have already operated all the axis in the correct direction and homed the machine successfully.
Are you homing the machine before you try probing Z

Kurdl, 2019-03-20:
Hi, MeJasonT!
I will repeat in English.
1. The machine responds correctly to all the commands of the programs: UGS, GRBL Controller, GrblGru, etc.
The machine correctly executes all NC.
The machine responds correctly to all commands of the "Candle" program.
The spindle moves down after pressing the "down" button.
Also, the spindle moves down after pressing the "Z-probe" button.
But after the command "Probe" of the heights map, the spindle moves upwards.
2. I didn't homing the machine before I try create the map.
But I started the search for the zero Z-position and it ended successfully ("Z-probe" button).

Kurdl, 2019-03-20:
My mistake: I did not reset the Z-axis before the start of sensing: (
I expected "Candle" to do this automatically.
I urgently need a manual for "Candle"!
Where to find it? :)

MeJasonT, 2019-03-20:
Unfortunately my friend we will have to write our own manual, to the best of my knowledge it has not been written yet.

Kurdl, 2019-03-20:
The Russians have a saying: "The salvation of the drowning is the business of the drowning themselves" :)
Maybe you, my friend, know the answer to another question:
Does Candle take $23 parameter (Inversion of the starting point direction, mask) into account to calculate the working coordinates?
How does this affect the "Restore Coordinate System" command?

MeJasonT, 2019-03-20:
I'm not 100 % sure but the GRBL/GNEA link is here with the information
[https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle](url)
J.

ondras12345, 2019-03-21:
> Does Candle take $23 parameter (Inversion of the starting point direction, mask) into account to calculate the working coordinates?
> How does this affect the "Restore Coordinate System" command?

Candle doesn't calculate the working coordinates. It only displays the value your machine sends.
Homing has effect to the machine coordinates and from my experience (grbl 0.9), it sets nonsense coordinates when I use the $23 parameter (I think it doesn't take the $23 parameter into account when setting the coordinates. But it should work if you home to the standard top right corner - the corner should become 0,0,0) Maybe this is different in grbl 1.1.
The restore coordinate system command resets the offsets set by G92 command (zero XY and zero Z buttons use the G92 command) http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g92

Kurdl, 2019-03-21:
After the homing cycle and resetting the machine coordinates, the working coordinates are not set to nonsense values ​​:)
Values ​​become equal to the parameters $130 - $132 minus $27 (Homing pull-off)

ondras12345, 2019-03-21:
> After the homing cycle and resetting the machine coordinates, the working coordinates are not set to nonsense values ​​:)
> Values ​​become equal to the parameters $130 - $132 minus $27 (Homing pull-off)

Ok, I've checked it and you are right. I expected it to make the bottom left corner 0,0,0 when I set $23=3.
Thank you