Denvi/Candle #273: Команда "Прервать" и её последствия.

Kurdl, 2019-04-03:
1. Отличная программа! Спасибо разработчикам.
2. Есть вопросы
Как осуществить не аварийное досрочное завершение УП?
Желаемый прецедент:
- нажимаешь "Стоп"
- станок останавливается, координаты сохраняются
- вызываешь возврат в нулевую точку и далее - по ситуации
Реальный инцидент:
Команда "Прервать" вызывает иногда просто катастрофические последствия.
А именно - скоростное перемещение инструмента в какую-то непонятную (и чаще всего разную) позицию.
Эта команда порождает цепочку кодов: G54 G17 G21 G90 G94 M9 T0
Что посоветуете?
Лучший ответ: "Читай мануал на Candle, который лежит по адресу http://... в главе такой-то всё написано".
Спасибо!

MeJasonT, 2019-04-03:
The operation you are referring to is controlled by GRBL and not Candle (the GUI).
the feed hold as an option in GRBL config.h Code to enable a safety door feature.
on triggering the safety door/feedhold the tool will extract and the spindle is shut down.  The current file position and work coordinates are held, when you select resume the machine goes back to its previous position and continues from that location.
I believe its possible to write additional functionality to have GRBL do more when feed hold is triggered

MeJasonT, 2019-04-03:
The operation you are referring to is controlled by GRBL and not Candle
the feed hold as an option in GRBL config.h Code to enable a safety door feature.
on triggering the safety door/feedhold the tool will extract and the spindle is shut down.  The current file position and work coordinates are held, when you select resume the machine goes back to its previous position and continues from that location.
I believe its possible to write additional functionality to have GRBL do more when feed hold is triggered