Denvi/Candle #275: Программа не открывается после первого запуска

ZombiePm, 2019-04-23:


ZombiePm, 2019-04-23:
После первого запуска программа вносит изменения в файл 
settings.ini
Эти значения
header=@ByteArray(\0\0\0\xff\0\0\0\0\0\0\0\x1\0\0\0\x1\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\x6\x30\0\0\0\x2\0\0\0\x5\0\0\0}\0\0\0\x4\0\0\0}\0\0\x2\xb9\0\0\0\x6\0\x1\0\0\0\0\0\0\x1\0\0\0\0\0\0\0}\0\0\0\x32\0\0\0\x84\0\0\0\0\0\0\0\x6\0\0\0\x63\0\0\0\x1\0\0\0\0\0\0\x1\x5\0\0\0\x1\0\0\0\0\0\0\0\x9b\0\0\0\x1\0\0\0\0\0\0\0\xb6\0\0\0\x1\0\0\0\x1\0\0\0\0\0\0\0\x1\0\0\0\0\0\0\0\0\0\0\0\x1\0\0\0\0\0\0\x3\xe8\0)
splitter=@ByteArray(\0\0\0\xff\0\0\0\x1\0\0\0\x2\0\0\x1\xd0\0\0\x1\x39\x1\0\0\0\f\x1\0\0\0\x2\0)
formGeometry=@ByteArray(\x1\xd9\xd0\xcb\0\x2\0\0\0\0\0\v\0\0\0\n\0\0\x3\xf5\0\0\x3u\0\0\0\x14\0\0\0.\0\0\x3\xec\0\0\x3l\0\0\0\0\0\0\0\0\a\x80)
Меняет на
header=@ByteArray(\0\0\0\xff\0\0\0\0\0\0\0\x1\0\0\0\x1\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\x6\x30\0\0\0\x2\0\0\0\x5\0\0\0}\0\0\0\x4\0\0\0}\0\0\x2\xd0\0\0\0\x6\0\x1\0\0\0\0\0\0\x1\0\0\0\0\0\0\0\x64\0\0\0\x32\0\0\0\x84\0\0\0\0\0\0\0\x6\0\0\0\x63\0\0\0\x1\0\0\0\0\0\0\x1\x5\0\0\0\x1\0\0\0\0\0\0\0\x9b\0\0\0\x1\0\0\0\0\0\0\0\xcd\0\0\0\x1\0\0\0\x1\0\0\0\0\0\0\0\x1\0\0\0\0\0\0\0\0\0\0\0\x1\0\0\0\0\0\0\x3\xe8\0)
splitter=@ByteArray(\0\0\0\xff\0\0\0\x1\0\0\0\x2\0\0\x1\xd0\0\0\x1\x39\x1\0\0\0\f\x1\0\0\0\x2\0)
formGeometry=@ByteArray(\x1\xd9\xd0\xcb\0\x2\0\0\0\0\b%\0\0\0\x37\0\0\f\r\0\0\x3\x9c\0\0\b-\0\0\0V\0\0\f\x5\0\0\x3\x94\0\0\0\x1\0\0\0\0\x5\0)
После чего приложение больше не открывается
если откатить эти изменения то программу можно откыть один раз

Denvi, 2019-04-24:
Посмотрите #173. Возможно, проблема в двух мониторах.

closed 2025-08-30 by github-actions[bot]