Denvi/Candle #277: trouble with heightmap, always 00.000

abdelabdel1, 2019-05-18:
hello
i have a trouble with heightmap:
The probe stop at the contact of cooper (pcb),  but the value that is written is always 0.
I have used heighmapp before and that always worked, i don't know what's happen.
thank you

eightheads, 2021-03-26:
I'm running into this same issue as well, all probe points on the heightmap read 0.000. Did you have any luck fixing this?
[image]

abdelabdel1, 2021-03-26:
I use an old version and that works

-------- Message original --------
Objet : Re: [Denvi/Candle] trouble with heightmap, always 00.000 (#277)
De : eightheads
À : Denvi/Candle
Cc : abdelabdel1 ,Author

I'm running into this same issue as well, all probe points on the heightmap read 0.000. Did you have any luck fixing this?
[Screen Shot 2021-03-26 at 3 11 32 PM]<https://user-images.githubusercontent.com/13207144/112690304-8c262900-8e49-11eb-9f39-5708d14dbfb2.png>

eightheads, 2021-04-06:
> I use an old version and that works

I came across this issue posted as well: https://github.com/Denvi/Candle/issues/341 and was able to get it working. You can either edit the hex in the executable file or edit source itself and re-compile.

abdelabdel1, 2021-04-07:
Thank you, but the new version is ok

-------- Message original --------
Objet : Re: [Denvi/Candle] trouble with heightmap, always 00.000 (#277)
De : eightheads
À : Denvi/Candle
Cc : abdelabdel1 ,Author

I use an old version and that works

I came across this issue posted as well: #341<https://github.com/Denvi/Candle/issues/341> and was able to get it working. You can either edit the hex in the executable file or edit source itself and re-compile.

eightheads, 2021-04-07:
> Thank you, but the new version is ok
Sorry, but that is incorrect, the code has not been changed. The main issue comes in if you are running GRBL with more than 3 axis as the current regular expression cannot handle more than 3 axis. It should be changed to ([^],:]*) to properly handle that going forward.

[image]

abdelabdel1, 2021-04-07:
Hello I have the 1.5.7.6 version and all is fine

-------- Message original --------
Objet : Re: [Denvi/Candle] trouble with heightmap, always 00.000 (#277)
De : eightheads
À : Denvi/Candle
Cc : abdelabdel1 ,Author

Thank you, but the new version is ok

Sorry, but that is incorrect, the code has not been changed. The main issue comes in if you are running GRBL with more than 3 axis as the current regular expression cannot handle more than 3 axis. It should be changed to ([^],:]*) to properly handle that going forward.
[Screen Shot 2021-04-07 at 11 10 03 AM]<https://user-images.githubusercontent.com/13207144/113899292-46932580-9792-11eb-9904-8485e5a0e769.png>

closed 2025-08-30 by github-actions[bot]