Denvi/Candle #278: Stepper motor doesn't work on CNC. Woodpecker board Candle

GissSH28, 2019-05-25:
Hi, I have a problem with my cnc, I buy a WoodPecker 2418 with the board. It was working well, but a few days i connect in my computer, the port its ok, but the stepper motors dont move. I thought could be the board, but when i tested the spindle works rightly. it Just the stepper motor i dont have response of them.  I prove it, one by one, and are okey. I used  CANDLE (GRBL CONTROL)

ondras12345, 2019-05-26:
Hi,
How did you test the steppers?
How are you trying to move them using Candle? Are you using the jog function?
Which version of Candle and GRBL are you running? (The GRBL version gets printed to the console when Candle connects to the machine)

GissSH28, 2019-05-26:
Yes, I trying to move them  using the jog function, and automatically too (with a nc file).  
The version of the GRBL is 0.9 and the candle is 0.8.4.
I had used them before and the CNC worked properly, but that was a year ago, stopped using it a long timeand now that i wantedtotest the setep motors doesn't responde, as i mentiones just the spindle.

FelisucoVFR, 2019-05-26:
Check the Windows language settings. Lately the steppers doesn’t move unless you set the US number settings in the language properties.
I had that issue when I change it to Spanish and stopped working.

GissSH28, 2019-05-26:
At the beginning, I had candle in  Spanish and used numbers in the metric system. Now I tried to switch it to English (as you comment) as well the imperial Unit system (since it's the one that appears to me) but I still have the same problem, the steppers doesn't move.
Also , I check the command window and in that moment that i put cnc in the HOME box is the following appears: 
[image]

ondras12345, 2019-05-27:
You have to configure homing if you want to use it. See https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9
I don't know why the steppers aren't working.

ondras12345, 2019-05-27:
You can try connecting to the machine using a terminal, eg. Putty or Hyperterminal and try typing the commands directly into the console. 
Use the same COM port settings as you configured in Candle.
Try:
G21 (mm mode)
X5 (move 5mm to the right)

You should set the terminal to send full lines (wait until you press enter) instead of sending the individual characters.

GissSH28, 2019-05-27:
> Check the Windows language settings. Lately the steppers doesn’t move unless you set the US number settings in the language properties.
> I had that issue when I change it to Spanish and stopped working.

In which part of the software do you change the language properties? To me candle appears totally in Spanish, and the only thing that could change with respect to language would be the system of units that would be from metric to imperial. You mean that?

FelisucoVFR, 2019-05-27:
Exactly, the Windows system.
The issue I had was the swap between the “,” and the “.” for decimal numbers.
It seems Now!

GissSH28, 2019-05-27:
I want to inform you that I already tried to change the region of the computer to see if this solved the problem, but no, the problem continues.

On the other hand, probe the motors individually to rule out the malfunctioning of these and for that reason they did not respond, but they are in perfect condition.

The only thing I have left to think is that the board is not working properly.  :(

ondras12345, 2019-05-28:
Did you try typing 
```
G21
X5
```
directly into the Candle console? It should move, even if it is the localization problem. Also, please check the "Show UI commands" checkbox in Candle settings, try jogging and post the text from the console.

GissSH28, 2019-06-01:
Hello I come to give reports and ask for opinions, I have verified that what does not work is the control board. Try the CNC with another control board equal to mine and effectively the CNC worked correctly. Verify that when you connect the power supply (on the control board if it works) turn on three LEDs, and when you connect the one that I see does not work, just turn on one. I attach the images so that someone can tell me the meaning of each LED. I wanted to try continuities inside the plate but apparently everything is fine. I do not understand the reason why it stopped working since it does not have two or three times that it was working.
[image]
[image]

ondras12345, 2019-06-02:
It looks like the 12v regulator is dead. Do you have a multimeter? Can you check some of the components? Focus on the part of the board labeled 12V DC power supply.
I'm suspecting the XL4015E1 IC is dead, you may need to replace it. Check if there is 24v on its input. If there's no voltage on its input, try to find a SMD fuse or a broken trace. If there is voltage on its input, there's something wrong with the power supply part.

GissSH28, 2019-06-07:
Hello he was doing tests and you are correct, the regulator is dead since it does not provide the 12v necessary for the movement of the motors. Do you think it is necessary to change the whole module (DO NOT THE REGULATOR FIND IT ONLY, ONLY IN MODULE) or just give the 12 VOLTS TO THE STEPPERS IN THE CORRESPONDING LINE?

ondras12345, 2019-06-07:
You can power the 12v rail from a separate power supply, just don't forget to disconnect the on-board (failed) regulator (you can leave the GND connected but disconnect output (pin sw - the one that goes to the inductor) and 24v from the chip. It might be easier to desolder the whole chip). I don't know how much current the steppers need, so you should pick a power supply that has the same or higher maximal output current as the original.

closed 2025-08-30 by github-actions[bot]