Denvi/Candle #280: Candle working under window 10

brianhomer, 2019-06-14:


brianhomer, 2019-06-14:
Candle under window 8 operation is fine, However, under  window 10 might has a problem, to fix it, you need the following’

1.	You need the compatibility check which is right click mouse on candle icon the number 2 below the administrator authority
2.	After checking choose trying suggestion for setup
3.	Then use the testing button, and save the setting.

McLovin42069, 2020-01-11:
I am having the same problem after switching to windows 10, but I have no idea what the candle "icon" is, or the number two below the administrator authority. Also, what do you mean by choose trying suggestion for setup. Thank you I greatly appreciate any help from anybody!!!!

ondras12345, 2020-01-11:
> I am having the same problem after switching to windows 10, but I have no idea what the candle "icon" is, or the number two below the administrator authority. Also, what do you mean by choose trying suggestion for setup. Thank you I greatly appreciate any help from anybody!!!!

I _think_ he means the Windows compatibility mode (see [the first result of google search](https://www.computerhope.com/issues/ch001894.htm)).

closed 2025-08-30 by github-actions[bot]