Denvi/Candle #282: grblcontrol no longer work graphics. Heeelp

Darkwolf77, 2019-06-22:
Hi.
I do not know what to do. I have a Woodpecker GRBL 0.9 cnc. I had worked with her without problems, one or two years ago. But now, the Grblcontrol does not show the graphics, it's the same computer. I have tried updating the drivers, reviewing the OpenGL version (it is 4.6), I have downloaded all the versions of Grblcontrol and in all the graphics do not work, I can control the cnc but blindly. please someone who can help me.

MeJasonT, 2019-06-22:
if candle opens and runs but its simply a case that the tool graphic is not displaying then have a look at your grbl $10 setting.  The status report has changed in the latest versions of grbl and now transmits the full report on $10=3.

Darkwolf77, 2019-06-22:
ok, thanks for answering, but in Candle, I can not manually control the cnc, it recognizes and connects, but I can not control it manually. he does not show me the graphics either, I entered $ 10 = 3 on the console, and he says "ok", and that's it, it's still the same :( maybe I'm an idiot and I'm not doing something right.

MeJasonT, 2019-06-22:
Hey buddy - your no idiot don't say such things.
It may be a case of the grbl version loaded on your machine.
I was under the impression candle was tolerant of the grbl version change from grbl 0.9 to 1.

Is your grbl board responding to commands sent them from the arduino ide instead - just to prove your hardware is working correctly and your communications connection is working.

try connecting with grbl-panel another gui like candle
https://github.com/gerritv/Grbl-Panel/releases

When did you reinstall candle, if you did it today and didn't reboot your PC it does odd things like this.

Darkwolf77, 2019-06-22:
okay. I have tested the grbl-panel, it recognizes the hardware (it is a woodpecker cnc grbl 0.9 card), the manual commands work, it loads a g-code .nc file and I render it without problems, but the speed of rotation of the motor was very very low (I imagine that you can set the speed of rotation from the console, but I do not know how and I did not find a graphic display of the code). I have restarted the pc, opened the candle again and it remains the same, it does not show the graphics, and it does not allow me to use the manual controls, but when loading the g-code it runs without problems. All other versions of the grblcontrol 0.7, 0.8, 0.81, 0.84, recognize and connect with the hardware, I can use the manual commands, and run the g-code without problems, but do not show the graphics. As I said before, I do not know what to do or how to solve this ...: (... and thanks, I'm not really an idiot, but these situations make me suspect otherwise xD ...

MeJasonT, 2019-06-22:
when you mention old versions of grbl 0.7, 0.8, 0.81, 0.84. i suspect your woodpecker board is Pre version 1.0 grbl.  The latest version of grbl has a hardware change with the variable spindle and z limit switch being swapped - it may be worth your time searching google to see if its a known issue and how others have worked around the old version running with woodpecker and candle.

answer appears to be here
https://www.cnczone.com/forums/chinese-machines/348746-cnc-forum.html

same issue
https://github.com/Denvi/Candle/issues/274 - nice person

two useful links
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

i will be away from pc for a while but will get back to you later if you are still having issues

Darkwolf77, 2019-06-22:
okay. The version of the controller card is, Woodpecker cnc grbl 0.9j, I have downloaded and tested the versions of Candle 0.9, 1.0.10 and 1.0.11, they work the same as the others, they work, but my problem is not with the limits or with the configuration of the grbl card (so far), my problem is with the program grblcontrol or candle, or the programs, because none shows me the 3D images where you can see the progress of the work virtually in the panel "G -Code Program ", does not show image or coordinates, I can only see the position icons to toggle the image, I do not see the design of the g-code nor the arrows x, and, z the image of the cutting tool. I have win10, Opengl 4.4, original drivers already changed, and I still can not see the 3D images ....: (... I want to cry ... xD, any idea of how I can solve this?
[image]
[image]

MeJasonT, 2019-06-22:
I realised what you were saying with regards the missing graphics and was trying to establish if you were getting information from the board or not, following a methodical fault finding approach.
then you go and throw windows 10 in the mix - there is your problem right there, its cack.
There is a report regarding a win 10 fix here
https://github.com/Denvi/Candle/issues/280, is this something close

Darkwolf77, 2019-06-22:
I have tried the suggested solution #280 in all three versions of Candle and there are no changes :(

Darkwolf77, 2019-06-23:
okay. In safe mode, all the versions of Candle work, but obviously they do not connect with the hardware, but if you can see the 3D images, now, that will be what happens in safe mode if you see the 3D images but in normal start not ...
[image]
[image]

Darkwolf77, 2019-06-23:
install the computer drivers and everything remains the same ...

marsheng, 2021-12-20:
I have the same problem. I have 2 PC with Win7. One works fine and I see the G-Code paths. The other, nothing. It runs the CNC machine correctly so everything is talking. 

On the second machine, I did a fresh update and all ok but still nothing. It would seem that the graphics has a problem. On this machine, the Visulation screen is completely blank. No O.O XYZ positions no timer no buffer etc. 

Really keen to get this working but don't know where to look.

marsheng, 2021-12-20:
I tried LaserGBRL on the same machine and it displays the G-Code so something wrong in Candle. .

marsheng, 2021-12-23:
I think I have found the problem. The program needs OPENGL 2 or above. The older machine is only1.1. My other is 3. something.

closed 2025-08-30 by github-actions[bot]